mne-tools/mne-python #1730: ENH: show whitening in Report if cov and evoked are found

dengemann, 2015-01-01:
cc @mainakjas

Eric89GXL, 2015-01-02:
Sometimes multiple covariance matrices (empty room, baseline) and evoked
files (multiple files with multiple conditions collapsed differently) can
be included in a report, so we have to figure out how to handle that
gracefully. +1 with that caveat in mind

agramfort, 2015-01-02:
I would add a --cov param like with we have a --info to specify the cov to
use.

mainakjas, 2015-01-02:
ok, noted. I will take care of this slightly later :)

Eric89GXL, 2015-03-09:
@mainakjas do you have time for 0.9?

agramfort, 2015-03-09:
@mainakjas arrives on friday in Paris for his PhD so he will soon have more
time :)

mainakjas, 2015-03-09:
indeed :)

Eric89GXL, 2015-04-20:
@mainakjas do you have time for this now?

mainakjas, 2015-04-20:
tomorrow!

Eric89GXL, 2015-04-20:
Cool, will leave marked as 0.9, then. Thanks for helping push us toward release.

closed 2015-04-27 by Eric89GXL