mne-tools/mne-python #1734: FIX/STY: avoid overwriting unbuilt python functions

dengemann, 2015-01-06:
In many places we've got things like `format` inside functions or elsewhere. We should avoid overwriting builtins alltogether.

wmvanvliet, 2015-01-08:
`format`'s a build-in? I though it was just a method of the base libs (as opposed to `print`, which is a build-in).

dengemann, 2015-01-08:
Yes it is :) you can try it :) it's the functional equivalent of the method. And it's indicated in -- guess -- all Python syntax schemes accordingly. Compare colors between e.g. `format`, `str` and `np.array`.

wmvanvliet, 2015-01-08:
so it is… learn something new every day :)

Yeah, we should not overwrite build-in names...

> 
> Yes it is :) you can try it :) it's the functional equivalent of the method. And it's indicated in -- guess -- all Python syntax schemes accordingly. Compare colors between e.g. format, str and np.array.

dengemann, 2015-01-08:
PR welcome :)

TanayGahlot, 2015-01-25:
So, What do you propose to use instead?

agramfort, 2015-01-25:
format -> fmt 

fmt is used in numpy

TanayGahlot, 2015-01-26:
To resolve this issue i would have to change all the format occurrence to fmt in the source code, can i take it up?

agramfort, 2015-01-26:
go for it.

TanayGahlot, 2015-02-14:
can you please elaborate on "fmt in numpy"?

mainakjas, 2015-02-14:
I think what @agramfort is referring to is the conventions followed in numpy. For example, here: https://github.com/numpy/numpy/blob/master/numpy/core/records.py#L275, they use `fmt` instead of `format`. We should do similarly in our repository. Makes sense?

TanayGahlot, 2015-02-14:
okay, i get it. Thanks :-)

TanayGahlot, 2015-02-15:
So if its 
```python 
'{0} ({1})'.format(event_id_rev[ev], np.sum(ev_mask))
```
I have to convert it to
```python 
'%s (%s)'%(event_id_rev[ev], np.sum(ev_mask))
```

agramfort, 2015-02-15:
you did not get it. It's otherway around. Sometimes we use format not for
string fomatting. See in bti/read.py for example. It's these usage of
format that are wrong.

TanayGahlot, 2015-02-15:
oh! right coz it overrides the inbuilt.So wherever format is used in similar context as read.py, i have to change that to fmt. Hopefully now i get it.

TanayGahlot, 2015-02-15:
I have sent PR #1800. Kindly review.

PR #1946 (2015-04-11): FIX: Dont shadow builtin
Closes #1734.
closed 2015-03-14 by teonlamont