mne-tools/mne-python #1737: ENH: automatic bad channel / bad epoch detection

wmvanvliet, 2015-01-10:
On a dataset with many subjects and many epochs, eye-balling data to mark bad channels and bad epochs becomes cumbersome. There are some papers out there that describe some statistics that can serve to automatically tag bad data, such as:

H. Nolan, R. Whelan, R.B. Reilly: FASTER: Fully Automated Statistical Thresholding for EEG artifact Rejection
http://www.sciencedirect.com/science/article/pii/S0165027010003894

It boils down to calculating things like the variance and correlation between channels and mark anything with a standard deviation > 3.

This would be in addition to manual checking of course.

dengemann, 2015-01-10:
Hey, take a look at mne.preprocessing.bads 

It goes in that direction. With @fraimondo we do something similar locally. Maybe we can stick our heads together and extract a contrib

dengemann, 2015-01-10:
Here: https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/bads.py#L9

Currently only used with ICA. It's a bit hidden but we use the same function already as part of a bad segments and bad channels detection workflow. Needs a bit more dev effort to make it generally usable. I'd be +100 at least for defning an MNE way of automatically detecting bad channels. In my experience the threshold of 3 std is not necessarily easy to motivate across datasets. The trick here is to drop outliers and recompute the std on the remaining channels / features. Usually we use 5. But this is not fully general. 

These examples might alos be relevant / interesting:
 http://scikit-learn.org/stable/modules/outlier_detection.html

dengemann, 2015-01-10:
And we have some open issue for exactly this quesion. We should find and close it in favor of this one.

wmvanvliet, 2015-01-10:
That's a good start.

Detection should preferably operate on Epochs, because during pauses, the subject might move / blink / rub eyes / speak / etc. causing the signal to be abnormal.

dengemann, 2015-01-10:
> Detection should preferably operate on Epochs, because during pauses, the subject might move / blink / rub eyes / speak / etc. causing the signal to be abnormal.

@wmvanvliet you see why we wanted the referencing there :) that's exactly the reason. Also note that we compute ICA on epochs for EEG.

wmvanvliet, 2015-01-10:
sklearn has fancy outlier detection... maybe a bit too fancy for what we need. The `find_outlier` function should serve as a good beginning. 

I'm playing with a testing dataset that contains a known bad channel. Feeding the vars of the channels through this function finds it.

dengemann, 2015-01-10:
> I'm playing with a testing dataset that contains a known bad channel. Feeding the vars of the channels through this function finds it

that works already pretty well with MNE sample.

But it's too optimistic where the bad channel changes over time and you've got many bad channels.

wmvanvliet, 2015-01-10:
We could process the data in chunks of maybe a minute. Channels may only be bad at certain times (electrode get loose halfway through the recording). An alternative approach is to mark bad channels for each epoch, although that may be a bit short time-window. Also, things like eye-movement correction should preferably be done before this analysis. The frontal electrodes in my sample set have a tendency to show up as bad, due to eye-blinks...

dengemann, 2015-01-10:
> We could process the data in chunks of maybe a minute. Channels may only be bad at certain times (electrode get loose halfway through the recording). An alternative approach is to mark bad channels [...]

@fraimondo WDYT?

agramfort, 2015-01-11:
how can you setup a large scale experiment to validate an awesome automatic
bad channel detection method?

wmvanvliet, 2015-01-11:
Yes and no. I have around 60 EEG recordings and around 30 MEG+EEG recordings. Any method we can come up with can be tested on those. However, I don't have time for a true study with accompanying paper. That would be a good master thesis project.

agramfort, 2015-01-11:
open problem? how do we share 200 or 300 subjects recordings across centers
to make such a study possible?

dengemann, 2015-01-11:
@fraimondo we should really think about sharing insights form our preprocessing routines here. We validated it on more than ~150 EEG recordings.

wmvanvliet, 2015-01-11:
That's mostly a political problem. My professor would never allow it.

agramfort, 2015-01-11:
> That's mostly a political problem. My professor would never allow it.

I know... :(

wmvanvliet, 2015-01-11:
We could approach it as a machine learning competition. We have our own datasets for local testing (training set). Once a method has been developed, it can be shared. Others can in turn try it on their data (test set) and report back the score.

agramfort, 2015-01-11:
hum a kind of distributed kaggle... interesting.

wmvanvliet, 2015-01-22:
First step of the FASTER algorithm:

https://gist.github.com/wmvanvliet/d883c3fe1402c7ced6fc

dengemann, 2015-01-22:
@wmvanvliet cool pretty much the same what I did locally. Let's make a branch for that.

agramfort, 2015-01-23:
+1 for WIP PR if experiments are conclusive.

dengemann, 2015-01-23:
@agramfort with @wmvanvliet we're on it, step by step.

wmvanvliet, 2015-01-28:
@dengemann Check out the gist. Second step of FASTER (bad epoch detection) implemented.

agramfort, 2015-01-28:
how do you set the thresholds? is it robust?

wmvanvliet, 2015-01-28:
the paper uses 3 std deviations as threshold. Robustness of this parameter needs to be determined using a large collection of recordings. I will attempt to do this after the FASTER algorithms has been completely implemented.

agramfort, 2015-01-28:
let's make a biggish validation database open some day...

wmvanvliet, 2015-01-28:
I found a collection of 109 EEG datasets:
http://www.physionet.org/pn4/eegmmidb/
That could serve as a good starting DB to test robustness.

agramfort, 2015-01-28:
sounds good. We need dataset fetchers for these.

wmvanvliet, 2015-01-28:
and we need to go over them manually, selecting bad channels, epochs, ICA components, etc... that will be a pain.

wmvanvliet, 2015-01-28:
updated the gist to contain an attempt at the third step (ICA) of the FASTER algorithm. This step really needs a second pair of eyes on it (@dengemann if you please?). I'm not that familiar with the semantics of performing ICA with MNE-python.

dengemann, 2015-01-28:
> and we need to go over them manually, selecting bad channels, epochs, ICA components, etc... that will be a pain.

what? why? not doing so is exactly why we want an automatic procedures. Yes @wmvanvliet I'll take a look in a bit.

wmvanvliet, 2015-01-28:
but the automatic method needs to be verified in some way. How will we do that without labeled data?

> 
> and we need to go over them manually, selecting bad channels, epochs, ICA components, etc... that will be a pain.
> 
> what? why? not doing so is exactly why we want an automatic procedures. Yes @wmvanvliet <https://github.com/wmvanvliet> I'll take a look in a bit.

dengemann, 2015-01-28:
But do you trust human observers ;-)
We could label a few or search for an independent criterion, e.g. compute
final SNR or some other statsitic.
At least we won't have the time to go through 100 files.

wmvanvliet, 2015-02-03:
good point :) What did the authors of the FASTER paper use?

Anyway, I've spend the day testing the 3 steps implemented in the gist. There were some bugs, which I fixed now.

agramfort, 2015-02-03:
does it work well out of the box?

wmvanvliet, 2015-02-04:
It can work really well, but you have to get the workflow right, which is not trivial. I believe the way forward with FASTER is not to present it as a boxed function that magically gets rid of all your artefacts. Instead, this gist will probably lead to 3 or 4 functions in `mne.preprocessing.bads` that are useful at various stages.

Here are my conclusions from my day of testing (sorry for the long read):

## Step 1: remove bad channels
At this step, channels that are hopelessly bad are noted in `info['bads']`. I've seen channels go bad due to the following reasons:

1. Sensor disconnected from system: flatline
2. EEG electrode loses contact with scalp: huge 50Hz powerline noise
3. EEG electrode's cable is caught somewhere, as the subject moves, so does the electrode: huge amplitudes in the signal

An effective test in all is these cases is simply `var(x)`. Additionally, testing for power at 50/60Hz is extremely effective in detecting loose channels. I haven't come across a case where bad channels are not picked up by the combination of these two tests.

However, there are 3 problems:

1. If the subject is taking a break and stretches muscles or talks, amplitudes for all channels increase hugely, making the tests less effective.
2. If lowpass filtering has been applied (<50Hz), the tests often fail to detect loose electrodes.
3. If the threshold for rejection is set too low, channels with a lot of EOG activity are sometimes flagged as bad.

Therefore, I found the most effective place for this test is to have epoched data (no rest periods for the subject), that has been highpass filtered, but not lowpass-ed, no average referencing, with the threshold set to 5 standard deviations. Under these conditions, the tests work beautifully.

## Step 2: remove bad epochs
At this step, epochs are discarded that are beyond hope of repair. I've seen bad epochs due to:

1. Subject moves
2. Subject bites down, yawns or talks (yaw muscles are strong!)
3. Large DC drifts not corrected by the highpass filter

The combination of `mean(var(x))` and mean-channel-deviation caught all above cases, if the threshold is not too high. I found the most effective place for these tests is after average referencing, before further bandpass filtering (only highpass), with a threshold set to 3 standard deviations. Under these conditions, the tests successfully reject really bad epochs. However, epochs where only 1 or 2 channels behave weird are not discarded. If no further preprocessing is done, a test should be added here that discards an epoch even if only one of the channels exceeds a threshold. This is currently implemented as the reject parameter of the Epochs object, a feature that could be improved by basing it on the outlier detection instead of a  fixed value.

## Step 3: remove bad ICA components
This step can save a lot of epochs that would otherwise have to be discarded, so it is probably important to get this right. This is where the FASTER algorithm implements the EOG correction, among other things. Some bad ICA components I've come across:

1. EOG artefacts
2. Alpha rhythm activity
3. 50Hz line noise

Unfortunately, the FASTER tests don't perform so well. The only components that ever get flagged as bad are due to the EOG correlation metric. And even EOG correction is not that good. It's better to first calculate the evoked potential using MNE's eye blink finder and then use SSP or ICA to calculate a good filter for EOG correction. All in all, I don't think FASTER has anything to add here that is not already implemented better in MNE-python.

agramfort, 2015-02-05:
on what data did you experiment this?

wmvanvliet, 2015-02-05:
some EEG data I have laying around. Now, you can test it yourself on the MNE sample data:
#1777.

Closing this in favour of the PR.

closed 2015-02-05 by wmvanvliet