mne-tools/mne-python #1738: ENH: maybe add pytoolz to externals

dengemann, 2015-01-11:
https://github.com/pytoolz/toolz/tree/master/toolz

https://toolz.readthedocs.org/en/latest/api.html#itertoolz

nice functional iterator patterns that help keeping the utils.py small.

agramfort, 2015-01-11:
what would you take from it?

dengemann, 2015-01-11:
Need to explore it a bit further. But almost everything it has seems pretty beautiful. Almost a bit sad this is not the standard library :(

dengemann, 2015-01-11:
certainly a few things that would help cleaning up control flows and container manipulations:
- concatnenation of sequences `concat` instead of the sum trick or comprehension tricks, also `mapcat`
- `groupby` seems pretty useful, maybe also for channel-wise iteratation
- `sliding_window`
- `partition_by`
- `pipe`, `compose` and `juxt`
- `do` for explicitly handling inplace modifcations
- `get` would also save many checks and apply to any python container

agramfort, 2015-01-11:
ok some maybe but beware of magic for new contributors.

dengemann, 2015-01-11:
not urgent but we should keep this in mind and see where it can save us lines and help make our code even more boring :) (wich is good)

Eric89GXL, 2015-04-12:
I'm -0.5 on using these for now. Perhaps we have to put in a little bit more effort currently to get things done, but it may be slightly more explicit / less magical (instead of having to look into what some function does to really understand). I'm going to go ahead and close, feel free to re-open if you find a compelling use case :)

closed 2015-04-12 by Eric89GXL