mne-tools/mne-python #1741: FIX: mayavi imports tend to produce x server crashes on some mashines.

dengemann, 2015-01-13:
would be good to avoid trying mayavi imports by option

cc @fraimondo @mainakjas @agramfort

agramfort, 2015-01-13:
how come it crashes?

dengemann, 2015-01-13:
segfaults via SSH

fraimondo, 2015-01-13:
It's not a segfault. It simply prints an error: "X server not found" or
something like that and then quits python.

And that happens when mayavi is imported. So all the code in one report
that tries to import mayavi does not work if mayavi is installed on the
system but the x server is not running.

It's a huge deal for distributed processing.

dengemann, 2015-01-13:
This should be fixed -- because it will lead to crahses irrespective of
whether you want use Mayavi at all

agramfort, 2015-01-13:
why not doing "ssh -X" ?

fraimondo, 2015-01-13:
Cause on a distributed environment you don’t have X servers. And sometimes you don’t want to have an X server running. That will increase too much the time to setup the environment.

Simple test to see the error:

In [1]: from mayavi import mlab
ipython: cannot connect to X server

agramfort, 2015-01-13:
see :

https://github.com/mne-tools/mne-python/blob/debian/debian/rules

for debian we use xvfb to fake the x-server.

would that help?

dengemann, 2015-01-13:
No it does not help, especially with xvfb-run this poses problems

fraimondo, 2015-01-13:
We are currently using that, although it seems that it has some huge memory leaks.

https://bugs.eclipse.org/bugs/show_bug.cgi?id=322108 <https://bugs.eclipse.org/bugs/show_bug.cgi?id=322108>
https://bugzilla.redhat.com/show_bug.cgi?id=498357 <https://bugzilla.redhat.com/show_bug.cgi?id=498357>
https://issues.apache.org/ooo/show_bug.cgi?id=105191 <https://issues.apache.org/ooo/show_bug.cgi?id=105191>

Workarounds that I need to test:
http://blog.jeffterrace.com/2012/07/xvfb-memory-leak-workaround.html <http://blog.jeffterrace.com/2012/07/xvfb-memory-leak-workaround.html>

Nevertheless, if we don’t really need mayavi it would be great not to use xvfb-run.

agramfort, 2015-01-13:
can you uninstall mayavi on the cluster? or have a venv without mayavi?

dengemann, 2015-01-13:
We'll subclass Report and overwrite the methods that are a problem.

fraimondo, 2015-01-13:
I’m currently doing some tests on that, to see if I can uninstall mayavi and everything works… will keep you updated.

fraimondo, 2015-01-14:
without mayavi:

Traceback (most recent call last):
  File "/opt/anaconda/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/opt/cpype/cpype/pipeline/tests/test_pipeline.py", line 219, in test_local_global_persistence
    pipeline.preprocess(raw, notch_filter=False, use_ctps=False, min_events=6)
  File "<string>", line 2, in preprocess
  File "/opt/anaconda/lib/python2.7/site-packages/mne/utils.py", line 526, in verbose
    return function(*args, **kwargs)
  File "/opt/cpype/cpype/pipeline/pipeline.py", line 481, in preprocess
    self._safe_figure(fig, 'Paradigm', section=section)
  File "/opt/cpype/cpype/pipeline/pipeline.py", line 354, in _safe_figure
    self.report.add_figs_to_section(figs, names, section, scale=scale)
  File "/opt/anaconda/lib/python2.7/site-packages/mne/report.py", line 785, in add_figs_to_section
    image_format=image_format)
  File "/opt/anaconda/lib/python2.7/site-packages/mne/report.py", line 720, in _add_figs_to_section
    if isinstance(fig, mayavi.core.scene.Scene):
UnboundLocalError: local variable 'mayavi' referenced before assignment

Eric89GXL, 2015-01-14:
That should be a fairly easy fix, just need to prepend a check for whether `mayavi` exists before moving on to the `isinstance` conditional, PR welcome :)

mainakjas, 2015-01-14:
mayavi is used only for rendering `-trans.fif` files. Why not use the pattern matching capabilities in report to render the `-trans.fif` files on a machine with X server capability and the rest of the files without the X server?

dengemann, 2015-01-14:
The problem is that even if you don't want mayavi at all you still deal
with it. It's imported in add figs to section to check for the type of
object. Maybe that can be avoided.

mainakjas, 2015-01-14:
yeah, we just need a check as @Eric89GXL says

agramfort, 2015-01-14:
+1 for the check. Report should work without mayavi installed.

Eric89GXL, 2015-04-10:
@dengemann would it be enough to just require that users for which `mayavi` will crash not have it installed, or do you still need a config option? Not a big deal either way, but it's simpler the former way.

closed 2015-04-11 by dengemann

tiandiao123, 2017-06-12:
may I ask how did you fix this problem? I has same issue.too.

fraimondo, 2017-06-24:
I don’t really remember.

I think I uninstalled mayavi, or commented out the problematic line in the server source code.

Eric89GXL, 2017-06-24:
You can uninstall, or edit `mayavi/__init__.py`` or `vtk/__init__.py` to have a `raise ImportError` as their first line if you for some reason don't want to delete them.

PR #1947 (2015-04-11): FIX: Safe not to have mayavi
Closes #1741.