mne-tools/mne-python #1742: ENH: use mne.cov sacaling functions in ICA

dengemann, 2015-01-13:
we should apply the same sort of scaling/rescaling everywhere. use mne.cov scaling functions

dlsrnsi, 2015-04-13:
Can I take it? But I see that mne.cov scaling functions are private functions..

agramfort, 2015-04-13:
if the problem is clear enough in your head go for it

dlsrnsi, 2015-04-16:
@dengemann which data should be scaled? ica.ica.unmixing_matrix_  or ica.noise_cov? or something else? And would you answer my question that mne.cov scaling functions are private functions.

dengemann, 2015-04-16:
They are private because they are not for the user but for the maintainers. But that does not really make a difference when using them in the code. This issue is actually not so easy, it takes some experience with ICA and this code to really evaluate it. The idea was to replace the standard scaling in the prewhitening function.

larsoner, 2021-02-19:
I think if people want to do this in practice they can just do `noise_cov=mne.make_ad_hoc_cov(info)`, so no need to implement this until people try it and see that it works better than the current approach which seems to be based on `std`. It will require at least a new parameter to enable the non-empirical / `scalings`-based mode.

closed 2021-02-19 by larsoner