mne-tools/mne-python #1743: ENH/FIX: use scipy / avoid creating figure for computing PSDs

dengemann, 2015-01-15:
mpl.psd does not play nicely with memory / forking / joblib. Use matplotlib.mlab.csd directly to avoid opening figures or use scipy.signal.welch instead.

example:

```Python

def _pwelch(epoch, nperseg, noverlap, nfft, fs):
    return [welch(channel, nperseg=nperseg, noverlap=noverlap,
                  nfft=nfft, fs=fs)[1]
            for channel in epoch]

def compute_psds(epochs, fmin=0, fmax=np.inf,
                 segment_size=128, n_overlap=100, n_fft=4096, n_jobs=1,
                 keep_mean=False, verbose=None):
    X = epochs.get_data()
    if keep_mean is not True:
        X = X.copy()
        X -= np.nanmean(X, axis=-1)[..., np.newaxis]

    sfreq = epochs.info['sfreq']
    parallel, my_pwelch, _ = parallel_func(
        _pwelch, n_jobs=n_jobs, verbose=verbose)

    psds = np.empty(epochs.get_data().shape[:-1] + (n_fft // 2 + 1,))
    freqs = np.arange(psds.shape[-1]) * (sfreq / n_fft)
    for i_epoch, fepoch in enumerate(parallel(
        my_pwelch(epoch, nperseg=segment_size, noverlap=n_overlap, nfft=n_fft,
                  fs=epochs.info['sfreq']) for epoch in X)):
        for i_channel, fchannel in enumerate(fepoch):
            psds[i_epoch, i_channel, :] = fchannel

    freq_mask = np.where(np.logical_and((freqs > fmin), (freqs < fmax)))[0]
    freqs = freqs[freq_mask]
    psds = psds[..., freq_mask]
    return psds, freqs
```

agramfort, 2015-01-16:
@yousrabk wan't to give it a try?

agramfort, 2015-02-21:
see #1748

PR #1914 (2015-04-02): ENH/WIP: use scipy for psds + spectral plots.
closes #1743 and #1856 .
closed 2015-02-21 by agramfort