mne-tools/mne-python #1749: ENH: improve epochs.plot (EEGLAB style plots)

dengemann, 2015-01-20:
Actuallys similar to plot raw, only show datasegments from epochs.

agramfort, 2015-01-20:
do you have a screen shot to illustrate?

dengemann, 2015-01-20:
Here (from the FASTER paper) used to select bad epochs.

[image]

Would be pretty easy and another conrtrib that removes more lines than it adds (recycling the plot_raw) ;-)

agramfort, 2015-01-20:
so just calling plot_raw on np.concatenate(epochs.get_data(), axis=0) ?
and adding the red colors?

dengemann, 2015-01-20:
... just adding the event markers instead. From the current  epochs.plot we still have the bad epoch slection tool. We should somehow transplant it then.

agramfort, 2015-01-20:
hum... so it's a plot_raw with highlighted bad segments? plot_raw can
already show events.

dengemann, 2015-01-20:
But this one only shows the epochs. not the rest. Epochs can be pretty remote from another. I think this is  better for data browsing than raw.plot with EEG data, which are often only viewable at that later stage, and it would be a nicer behavior for epochs.plot

agramfort, 2015-01-20:
ok sounds like a plan. PR welcome :)

Eric89GXL, 2015-12-12:
@dengemann do you still want this, or were the enhancements to Epochs plotting enough? I guess we could deal with this alongside #2665 potentially.

dengemann, 2015-12-12:
this was the GSOC with @jaeilepp !!

closed 2015-12-12 by dengemann