mne-tools/mne-python #1750: BUG in set_reference

dengemann, 2015-01-20:
```
In [14]: set_eeg_reference(raw)
Adding average EEG reference projection.
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-14-c07e9475d8ec> in <module>()
----> 1 set_eeg_reference(raw)

/Users/dengemann/github/mne-python/mne/io/reference.pyc in set_eeg_reference(inst, ref_channels, copy)
    175             return inst, None
    176         else:
--> 177             inst.add_proj(make_eeg_average_ref_proj(inst.info), activate=False)
    178             return inst, None
    179     else:

TypeError: add_proj() got an unexpected keyword argument 'activate'
```

cc @wmvanvliet

dengemann, 2015-01-20:
Also it would be nice to have this function in the mne namespace

wmvanvliet, 2015-01-20:
#1751

wmvanvliet, 2015-01-20:
@dengemann you mean `mne.set_eeg_reference` instead of `mne.io.set_eeg_reference`?

dengemann, 2015-01-20:
Yes! Or avoid that alltogether and add an instance method.

dengemann, 2015-01-20:
I was actually a bit surprised to find reference in io. Would make more sense to me in mne.channels. We could still change that without deprecation.

agramfort, 2015-01-20:
>
> I was actually a bit surprised to find reference in io. Would make more
> sense to me in mne.channels. We could still change that without
> deprecation.
>

yeah makes sense unless it creates circular deps

wmvanvliet, 2015-01-22:
I took the liberty of merging the bugfix. I'll see now about moving the function (less urgent).

closed 2015-02-21 by agramfort