mne-tools/mne-python #1753: Time indexing bug when converting epoch to dataframe

choldgraf, 2015-01-20:
I think I've found a bug when converting Epochs objects to pandas DataFrames using the .as_data_frame() method.

I found that when you try to reshape the resulting dataframe (e.g., df.unstack('time')) you get an indexing error saying that there are duplicates in the index. When I checked the "time" index, I found that all timepoints > 1 second were being duplicated. E.g. before 1 second, times were [997, 998, 999]. However, after 1 second, they become [1002, 1002, 1004, 1004, 1006, 1006] etc.

I reproduced this with the following code. I have an epochs object with trials > 2 seconds after onset:

    times = epochs.times * 1e3   # Using same conversion as default
    times.astype(int64)[(tst < 1006) * (tst > 996)]

This produces the following output for me:

    array([ 997,  998,  999, 1000, 1000, 1002, 1002, 1004, 1004])

Note that when I do this with the following code:

    times = np.arange(-.5, 2, .001) * 1e3`
    times.astype(int64)[(tst < 1006) * (tst > 996)]

Then it works fine, with output:

    array([ 996,  997,  998,  999, 1000, 1001, 1002, 1003, 1004, 1005])

So it looks like something is going wrong with rounding when converting to 'int64'. I tried it with int32 and the same problem occurs. 

Strangely, the duplication stops after 1025. E.g.:

```
array([ 997,  998,  999, 1000, 1000, 1002, 1002, 1004, 1004, 1006, 1006,
       1008, 1008, 1010, 1010, 1012, 1012, 1014, 1014, 1016, 1016, 1018,
       1018, 1020, 1020, 1022, 1022, 1024, 1025, 1026, 1027, 1028, 1029,
       1030, 1031, 1032, 1033, 1034, 1035, 1036, 1037, 1038, 1039, 1040,
```

Anyone else experience this?

dengemann, 2015-01-20:
@choldgraf hah, it seems you're among the few users that are using this. I've implemented it once but my predictions about the importance of pandas (and the stability of its API) did not come true, hence, this part of our code meanwhile became a bit neglected. We're actually discussing to remove these exporters. 

The error you're reporting probably depends on the version. Which version of pandas do you use? I think I've seen this before.

choldgraf, 2015-01-20:
Ah that's a total bummer! Pandas has become totally integral to my analysis pipeline and I use as_data_frame all the time :P ah well.

I'm on pandas 15.1

dengemann, 2015-01-20:
@choldgraf good you're raising your voice. I thought I was the only one using it.

dengemann, 2015-01-20:
I think our code is tested for pandas versions from 2012 to late 2013.

choldgraf, 2015-01-20:
I love pandas, makes plotting, playing around with data, etc much easier. If anything else, using indices so that you know exactly the details for each datapoint is invaluable to me.

Is pandas support that difficult? I'm surprised nobody else is using it!

Eric89GXL, 2015-01-20:
In my experience, getting pandas to work and continue working across versions, with their API changes and methods of data access and constructions, has always been a nightmare.

dengemann, 2015-01-20:
> Is pandas support that difficult? I'm surprised nobody else is using it!

it's really a pain, don't ask me where exactly. It was actually pretty challenging to implement code that covers the versions in the time range mentioned. Nothing like that can be said about e.g. numpy.

dengemann, 2015-01-20:
> . If anything else, using indices so that you know exactly the details for each datapoint is invaluable to me.

We can keep  this code if people are using it. But we'll need some active input/ contribs to keep it alive as I'm personally not using pandas any longer except for reading csv files.

choldgraf, 2015-01-20:
That makes sense if it's so tough to maintain. Sounds like people are starting to get jaded with pandas :)

I personally do use it a lot, but that's because I write a lot of custom code/analysis, and like to convert MNE objects into a more "raw" format in an intelligent way (e.g., as a data frame with the correct indexing). I suppose it's no so useful if most analysis stays within the scope of MNE-python.

If I'm the only other person that uses it, then maybe it sounds like it's too much of a hassle. Not sure how to figure out if it's something that's more common than just me...

dengemann, 2015-01-20:
> If I'm the only other person that uses it, then maybe it sounds like it's too much of a hassle. Not sure how to figure out if it's something that's more common than just me...

Well maybe you take over supplying this part of our code with some care ;-)
Let's track down the problem and see what we can do together.

choldgraf, 2015-01-20:
I'm certainly happy to help, though judging from your guys' last comments about it being a "nightmare", do you think it's something that a single person can realistically keep up with? That said, from my experience it seems that pandas is settling down a little bit as it matures in terms of API shifts and such. Maybe it'll be easier to support in the future?

dengemann, 2015-01-20:
> Maybe it'll be easier to support in the future?

that's a good view. I hope these guys will finally converge on a stable API regarding .index and related things.

dengemann, 2015-01-20:
Ok @choldgraf have you got a script that produces your problem? Could you propose a failing test?

Eric89GXL, 2015-01-20:
It very well could be easier to maintain now. I just remember a year or two ago, every new release meant digging through code with version-based conditionals to get things to work. IIRC some of my frustration also stemmed from poor API documentation related to the changes (and even basic access), which has also probably improved by now.

In any case, we should fix this error, since we provide the functionality we want it to work :) @choldgraf you up for giving it a try?

choldgraf, 2015-01-20:
yep - I can give it a shot. In this case, I don't think that pandas is the
problem. A kind of stupid solution that I just threw together is just to
re-create the time index using np.arange manually (e.g., use
np.arange(epochs.times[0], epochs.times[-1] + .001, .001)). Though it seems
that this is the same way that this is the same way .times is constructed,
so I need to do some more digging.

Either way, I can try to create code that will break, and can work from
there.

dengemann, 2015-01-20:
Either way, I can try to create code that will break, and can work from
there.

@choldgraf  exactly than let's find a modification that makes your new test pass but does not make the old tests crash + examples should work as they used to work before.

choldgraf, 2015-01-21:
OK, I'll give it a shot in the next day or so. Will be my first
contribution so apologies if it's a bit clunky.

choldgraf, 2015-01-21:
Quick question - for bugfixes / features / etc, do you want people to branch off of the latest maintenance branch, rather than master? AKA, I should branch off of maint/0.8, right?

dengemann, 2015-01-21:
@choldgraf we'll bother later about backporting, now let's first get it fixed on master.

choldgraf, 2015-01-21:
sounds good

closed 2015-04-12 by Eric89GXL