mne-tools/mne-python #1754: ENH: automatically determine fmin, fmid, fmax for plotting

wronk, 2015-01-21:
I always have issues setting fmin/fmid/fmax easily and thought it might be easier to have them set automatically based on the data distribution. It would be pretty easy to use numpy's percentile function so that those values could be set based directly on the data distribution (shout out to @Eric89GXL for the tip).

Would this be worth me implementing? If so, my main question is how/where should this functionality be included?

Eric89GXL, 2015-01-21:
There are many situations where I don't really like having to set `fmin/fmid/fmax`. Maybe we could deprecate their use and just have a `limits='auto' | tuple of length 3 | dict`, where the length-3 tuple is the old behavior, `dict` allows for doing something like `dict(type='percentile', bounds=[0.25, 0.5, 0.75])`, and `auto` is an alias for some suitable default percentiles.

agramfort, 2015-01-21:
fine with such API. shal we call it "flim"?

dengemann, 2015-01-21:
We have an open issue for that. Could you please find it, link it, carry over existing discussion and close it? This is a dupe.

dengemann, 2015-01-21:
see #1123

dengemann, 2015-01-21:
Here was an old API draft we discussed earlier
 
```Python
"""
threshold : float, int, tuple, dict,  'auto'
     The threshold for visualization. If `dict`, the keys 'fmin', 'fmid' and 'fmax' are required.
     If tuple, two ints or floats are required related to the lower and the upper percentile, fmid will  
     be set to zero in this case. If int or float, the percentile defining the threshold is expected.
     In this case 'fmax' will be set to 100 and 'fmin' will be placed at exactly the mean between the
     value chosen and 100.  
     If 'auto', depending on the sign of the data either defaults to '95' or to '(2.5, 97.5)'. In 
"""
```

Eric89GXL, 2015-01-21:
That seems like it would be close to working. One thing I don't get is what are the lower/middle/upper thresholds for the `(2.5, 97.5)` case? `(2.5, 50, 97.5)`? In other words, we have three control points (fmin/fmid/fmax), not two.

Also, we should have some way of dealing with whether the data should be considered unidirectional/unsigned or bidirectional/signed. For example, the `mne_analyze` colormap is good for data with positive and negative values, whereas the "fire" colormap is good for data with only positive values. When doing the percentile calculation, it might be important to take this into account -- if the data are signed, perhaps the percentiles should be on the `abs()` of the data and then a bidirectional colormap should be used.

dengemann, 2015-01-21:
> . For example, the mne_analyze colormap is good for data with positive and negative values, whereas the "fire" colormap is good for data with only positive values. When doing the percentile [...]

yes something like that. Or split percentiles for < 0 and > 0. One thing. We probably want to have some asymmetric thresholds depending on the distribution. Max should be 100, often 99 is still too low in my experience.

Eric89GXL, 2015-01-21:
I think the majority of use cases would benefit from using the same thresholds for both positive and negative, so I think by default they should match but have a kwarg to switch.

wronk, 2015-01-21:
I'm in favor of the default method taking the abs() when calculating the percentiles just to keep the function's behavior more straightforward/understandable. The more complex one will be easy enough to put into kwargs, or it could just check the flim dict for a `bounds` list that has either 3 or 6 values.

So far, all options I see so far for the percentile dict:
1) Single colormap or 
2) Bidirectional colormap where the cmap is 
  * mirrored across 0 (current behavior)
  * separate (different color scales for pos and neg values)

agramfort, 2015-01-21:
we can you a positive=True param to force positive colormap.

Eric89GXL, 2015-01-22:
Sounds like we have started converging on a solution -- maybe the best thing now is to implement, show a couple example code snippets (with plots) and we can evaluate / iterate from there.

wronk, 2015-01-22:
How about something like
`limits = 'auto' | (fmin, fmid, fmax) | dict(diverging, cmap_limits)`

`'auto'` (str) will use a diverging colormap with default percentiles
`(fmin, fmid, fmax)` (3-tuple of floats) will just replicate the current behavior
`dict(diverging, bounds)` (bool, list) will create either a single sequential cmap (think zero alpha to red) or a diverging cmap (think cyan to zero alpha to red) depending on the diverging flag. Then the bounds list containing percentiles can be length 3 or 6 depending on if the control points should be symmetric across zero or not. 6 points should probably only be allowed if diverging=True.

agramfort, 2015-01-22:
I don't find diverging very explicit but I like the idea.

wronk, 2015-01-22:
Easy change, so I'll get started.

@agramfort, what would you suggest? I think 'positive' might be a little ambiguous since all cmaps will have positive values... Other names often used to distinguish the two maps are bipolar and double-ended.

Eric89GXL, 2015-01-22:
`one-sided`?

agramfort, 2015-01-22:
> one-sided?

fine with me

wronk, 2015-02-03:
After some coding/talking with @Eric89GXL, here's the working API I'm imagining:

I'm still planning on using `limits = 'auto' | (fmin, fmid, fmax) | dict()`
The plan for the 'auto' and tuple hasn't changed. As far as the dict, it must contain enough info to:
1) indicate the colormap to be use (or construct it if it's an 'mne_analyze' cmap)
2) scale the colormap to the correct limits
3) indicate if the limits are hard number bounds, or if they are percentages

dict = {`kind` = str 'value' or 'perc|percent|percentage'
`lims` = len 3 list/array
`pos_lims` = None or len 3 list/array if using mne_analyze cmap
`neg_lims` = None or len 3 list/array if using mne_analyze cmap}

The `kind` flag indicates whether to use input numbers or convert those to percentages for the bounds.
`lims` will scale the colormap to the entered lims (when not using an mne_analyze cmap).
`pos_lims` will only be checked for mne_analyze colormaps and will mimic the traditional fmin/fmid/fmax behavior by mirroring these control points across zero.
`neg_lims` will only be for mne_analyze cmaps and allow for asymmetric control points in the mne_analyze cmap.

More suggestions?

Eric89GXL, 2015-02-03:
Note that the `limits` dict won't need to contain a colormap name, since there is already a `colormap` argument in the argument list. +1 for `kind='values' | 'percentages'`, I find that the most readable.

agramfort, 2015-02-03:
sounds ok to me. I am just not sure we want to support non-symmetric
colorbars with the pos_lims neg_lims. I'd rather just have lims.

Eric89GXL, 2015-02-03:
I'm fine with not having non-symmetric colorbars, but I think @dengemann wanted it so he might need some convincing. It does simplify the API a bit to force symmetry. I also think having the possibility of asymmetric colorbars is likely to cause more harm than good. So maybe we should leave it out for now. Okay with you @dengemann?

Even with non-symmetric colorbars, we need some way of distinguishing between limits for the `mne_analyze` colormap -- where the limits mean the three control points for the positive half of the colormap -- and the limits for every other colormap in the world -- where control points are just left, middle, and right. Otherwise we're stuck with the current way it has to be used, which is ugly.

For example, say I wanted a standard mne_analyze colormap with 5/10/15 limits for positive values, and -5/-10/-15 for negative, I'd do this:
```
stc.plot(..., colormap=mne.viz.mne_analyze_colormap([5, 10, 15]),
      limits=[-15, 0, 15])
```
The idea is that a user should instead be able to do something like this to get the proper behavior:
```
stc.plot(..., colormap='mne_analyze', limits=dict(pos_lims=[5, 10, 15]))
```
The "pos_lims" key would only be allowed for the mne_analyze colormap, or any other colormap where we actually have control over more control points than just left, middle, and right.

Also, we should change the default for `transparent` to be `None` and make it `True` (as it is now) for all colormaps except for `'mne_analyze'`, where it should default to `False`. Having it be `True` right now messes up the mne_analyze colormap because it causes the blue/negative half to alpha incorrectly.

agramfort, 2015-02-03:
please just keep it simple. A good exercise is to write the doc first to
realize if it's simple or not to explain :)

wronk, 2015-02-03:
Sounds good. @Eric89GXL , I'll code it up to include the `pos_lims` (to avoid overloading the lims parameter) and make sure it can be reasonably explained in the docstring but leave out the asymmetric behavior for the moment . I'll update the `transparent` param too.

@dengemann , we can definitely go through some auto thresholds as soon as I've gotten it functional. I think that will mostly come down to what data set is most representative of what users will be plotting.

closed 2015-04-10 by agramfort