mne-tools/mne-python #1758: Plot background image in topo plots

choldgraf, 2015-01-24:
It would be really useful (at least for ecog) to plot an image behind topo plots. This way, each plot would be an electrode, and you could see where these lie on the brain in the background. Perhaps it could be a flag to plot_topo functions/methods?

Eric89GXL, 2015-01-24:
It might be easiest / most general just to make sure the user can get to
the list of axes. Then they can add arbitrary other plots to each,
including background images. WDYT? My concern with adding an argument for
background images is that it seems pretty specific to a limited set of use
cases.

dengemann, 2015-01-24:
Calls for a `plot_ecog_topo` method

choldgraf, 2015-01-24:
Plot_ecog_topo could work. I wrote some code that takes a set of xy points
and transforms them into 0-1. That could work for ecog usecases.

I think you already have access to the axes objects via the figure. I just
couldn't figure out how to create a new axis that was behind all the other
ones. Tried looking for a z parameter, but could only find references to
within-axis z, not within-figure z.

Eric89GXL, 2015-01-24:
Adapted from http://matplotlib.1069221.n5.nabble.com/Control-twinx-series-zorder-ax2-series-behind-ax1-series-or-place-ax2-on-left-ax1-on-right-td12994.html
```
import matplotlib.pyplot as plt
ax1 = plt.subplot(211)
ax2 = plt.subplot(212)
ax3 = plt.axes([0, 0, 1, 1], axisbg='r')
ax3.set_zorder(min(ax1.get_zorder(), ax2.get_zorder()) - 1)
plt.show()
```
[image]

choldgraf, 2015-01-24:
Ah this looks great. OK, I'll try to make it work tomorrow, thanks for
digging that up! I guess that's why I shouldn't try to Google things after
debugging for 3 hours :-)

Eric89GXL, 2015-01-24:
I don't blame you -- tracking down that link felt like a google-fu stress test

choldgraf, 2015-01-24:
Haha yeah, interestingly Paul used to work down the hall from me...guess I shoulda just asked around :P

choldgraf, 2015-01-24:
OK so I got the image underlay working thanks to the link you sent. I probably can't post a picture of an ecog patient's brain on github, so I plotted it on top of batman instead. This is far more satisfying for me anyway.

[image]

I think that it will be tricky to plot in a manner that doesn't come off as way too busy. The image would need to be optimized for plotting lots of channels (or maybe it'd be better to only plot a subset of channels?).

Eric89GXL, 2015-01-24:
Sweet! Three things I can think of that would help make it less busy:

1. Mix the background with black (or better, give it an alpha and give the figure a black background), so it appears faded / darker.

2. Give the axes each have a black background, maybe with alpha.

3. Allow for selecting a rectangular region of the figure, and "zooming in".

Approaches 1 and 2 should be very easy. Approach 3 is cool but probably more work than it's worth, esp. given that these plots should already be click-able to open individual channels in new windows.

choldgraf, 2015-01-24:
Good call - before I was just using axis.patch.set_visible(False). If you
use set_alpha instead then it has the desired effect. I could also do the
same for the figure background.

Regarding crowding, I think that you're right - it'll be a fair amount of
work to make custom zooming functionality (at least for someone like me who
does not speak matplotlib-ese so well). That said, I think that we could
make it clear in the documentation that electrode images should be as
zoomed in as possible or as high res as possible.

Some quick questions:

1. I have a short python script that lets you open an image, click points
in that image, and then it stores the points in a dictionary. I've been
doing this to create custom ECoG layouts thus far, since we don't usually
have any information about electrode location other than the
reconstructions. Would this be in the scope of MNE or should we just assume
people will figure out their own way to get XY positions?

2. What kind of function would be shown to the user here? As Denis
suggested, we could include a "plot_ecog_topo" method that also accepts xy
positions as well as the epochs / evoked object, then automatically
generates a layout out of this and plots. Alternatively, there could be a
"generate_ecog_layout" function in the layout module that would let you
create whatever you want.

Chris

Eric89GXL, 2015-01-24:
I think that `generate_ecog_layout` or more generally `generate_2d_layout` or so would be more general, because then it can be generated once and saved, or passed to `plot_topo`. The issue of getting the appropriate XY positions could be general enough, and could even be part of the `generate_2d_layout`, like:

```
def generate_2d_layout(ch_names, pos='interactive', background=None):
    """Generate a 2D layout

    Parameters
    ----------
    ch_names ...
    pos : ndarray | str
        If ndarray, should be Nx2 array of positions. Can also be 'interactive'
        to allow interactive point selection by clicking within axes.
    bg_image : str | ndarray | None
        Background image to show during point selection. Can be None
        if no background is necessary. A str should be a filename to load,
        and an ndarray should be able to be plotted using mpl.imshow().
    """
```
That way the layout could be used directly with `plot_topo`... and I'm coming around to the idea of having a `bg_image` property of `plot_topo`, since it seems like it would be useful here, and would obviate the need for a separate `plot_ecog_topo` function.

agramfort, 2015-01-24:
+1

my brain is connected with Eric's with 9 hours of time lag :)

choldgraf, 2015-01-24:
What about something like:

```python
def generate_2d_layout(xy, w=.07, h=.05, ch_names=None, ch_indices=None, name='ecog', pad=.02, bg_image=None):
    '''Generates a 2-D layout for plotting with plot_topo methods and functions. XY points will be
    normalized between 0 and 1, where normalization extremes will be either the min/max of xy, or
    the width/height of bg_image.
    
    Parameters
    -------
    xy : ndarray (N x 2) | 'interactive'
        The xy coordinates of sensor locations. If 'interactive', bg_image must not be None, as this will be displayed
        and the user must click on positions of the image to create xy values.
        
    w, h : float
        The width and height of each sensor's axis (between 0 and 1)
    
    bg_image : str | ndarray
        The image over which sensor axes will be plotted. Either a path to an image file, or an array that can be
        plotted with plt.imshow. If provided, xy points will be normalized by the width/height of this image. If not,
        xy points will be normalized by their own min/max.
        
        
    Returns
    --------
    layout : Layout
        A Layout object that can be plotted with plot_topo functions and methods.
    '''
    # Check that len(ch_names) == len(ch_indices) if not None
    # Check that len(ch_names) == len(xy) if not None
    
    if ch_indices is None:
        ch_indices = np.arange(x.shape[0])
    if ch_names is None:
        ch_names = ['{0}'.format(i) for i in ch_indices]
        
    # To interactively pull xy values
    if xy == 'interactive' and bg_image is not None:
        # Pull xy coordinates interactively
    else:
        # Raise an error that we need an image to do interactive stuff
    
    # Normalize xy to 0-1
    if bg_image is None:
        # Normalize x and y by their maxes
        xy -= xy.min(0)  # Separate out into x and y
        xy /= (xy.min(0) - xy.max(0))
        y = 1-y # Flip y-axis
    elif bg_image == 'string':
        try:
            #imread image file
    elif isinstance(bg_image, np.ndarray):
        # Normalize by width / height of array

    # Create box and pos variable
    box = (0-pad, 1+pad, 0-pad, 1+pad)
    w, h = [np.array(i * x.shape[0]) for i in [[w], [h]]]
    loc_params = np.vstack([x, y, w, h]).T
    
    layout = mne.channels.Layout(box, loc_params, ch_names, ch_indices, name)
    return layout
```

agramfort, 2015-01-24:
sounds like a plan. Just a remark we call ch_indices picks for some
historical reasons.

choldgraf, 2015-01-25:
Sounds good - I'll try to throw something together. I was calling it
"ch_indices" because the current Layout class calls them "ids", which I
thought meant "indices". I can switch it to picks though. Should that be
called "picks" as well then?

agramfort, 2015-01-25:
we can rename later if necessary. Let's comment and improve on a PR.

closed 2015-04-28 by Eric89GXL