mne-tools/mne-python #1761: ENH: support joblib backend API and allow for 'threading'

dengemann, 2015-01-25:
Depending on the python setup and the data inputs spawning / forking may not be the best choice.
We should allow for Parallel(backend='threading').

dengemann, 2015-01-25:
We should explore this a bit before implementing it. As far as I can tell we gain substantial speedups. However, so far only tested in a context where increasing the number of nobs actually slows things down ...

Eric89GXL, 2015-01-25:
Where can I read about this? Does it allow sharing memory instead of
serializing it to each worker?

agramfort, 2015-01-26:
yes but for best efficiency you need cython gil free code.

Eric89GXL, 2015-04-12:
Do we have any code that releases the GIL? Maybe something that uses `sklearn` somewhere, like the machine learning code? Even then I'm not sure we'd release it enough to be effective. I think we can probably close this.

closed 2015-04-12 by agramfort