mne-tools/mne-python #1765: bug in read forward

agramfort, 2015-01-28:
The normals in fwd do not match when using fixed_ori or not. See below.

@Eric89GXL do you mind having a look?

```
import numpy as np

import mne
from mne.datasets import sample
from numpy.testing import assert_array_almost_equal

# Set path to forward solution
data_path = sample.data_path()
fwd_fname = data_path + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'

# Read surface oriented forward solution with free oriention
fwd_free = mne.read_forward_solution(fwd_fname, surf_ori=True,
                                     force_fixed=False)

# Read forward solution with fixed oriention
fwd_fixed = mne.read_forward_solution(fwd_fname, surf_ori=True,
                                      force_fixed=True)

ori_source1_fixed1 = fwd_fixed['source_nn']

assert_array_almost_equal(fwd_free['source_nn'], ori_source1_fixed1)
```

joewalter, 2015-01-28:
Hi Alex, thanks for opening this issue.

I think the problem is that for the rotation to surface orientation, one can use the patch average, while this option is not used currently when computing a gain matrix with force_fixed=True.

An option for solving this issue would be to compute the fixed orientation gain matrix by first computing a free orientation, surface-oriented gain matrix and then use only the rows of the gain matrix corresponding to the normal direction.

joewalter, 2015-01-28:
If you are interested I can open a pull request. The problem is that we will probably break the compatibility to MNE-C using this modification.

agramfort, 2015-01-28:
just use mne.forward.convert_forward_solution and your problem is solved.

fwd_fixed = mne.forward.convert_forward_solution(fwd_free, force_fixed=True)

close if you agree

joewalter, 2015-01-28:
I tested it with the code above and it doesn't solve the problem. 

In the current implementation, the source orientation in the force_fixed case is taken from fwd['src']. 
```
fwd['source_nn'] = np.concatenate([s['nn'][s['vertno'], :] for s in fwd['src']], axis=0)
```

For a free orientation forward with surf_orient, the 'nn' is computed from the patch_info if available:
```
if use_ave_nn is True:
    nn = s['nn'][s['pinfo'][s['patch_inds'][p]], :]
    nn = np.sum(nn, axis=0)[:, np.newaxis]
    nn /= linalg.norm(nn)
else:
    nn = s['nn'][s['vertno'][p], :][:, np.newaxis]
```
As loading a surface oriented forward will not change fwd['src'], the orientations will still be different if average patch normals are employed for the surface oriented free forward

agramfort, 2015-01-28:
if you convert to fixed a fwd with patch_info you keep the patch info right?

joewalter, 2015-01-28:
The force_fixed option doesn't care about patch_info. It takes the orientation from fwd['src'].

agramfort, 2015-01-28:
hum... how do we fix it then? Sorry I am slow...

joewalter, 2015-01-28:
The idea would be to always run the surface orientation code, so we check if patch info is available and compute the new local coordinate system. Then in order to force_fixed we just crop the respective rows/columns of source_nn and data.

Eric89GXL, 2015-01-28:
I'm not sure what the best solution is, but it does sound like taking the
patch info into account should work better. We should preserve the option
to maintain compatibility with the c code, though. Doing so should also
stay the default for at least one release via deprecation.

agramfort, 2015-01-28:
>
> The idea would be to always run the surface orientation code, so we check
> if patch info is available and compute the new local coordinate system.
> Then in order to force_fixed we just crop the respective rows/columns of
> source_nn and data.
>
please git it a try and see what it takes to do this. I'll comment on PR.

Eric89GXL, 2015-01-28:
@mshamalainen any reason not to take the patch information into account to compute the normal directions when making a fixed-orientation forward solution?

dgwakeman, 2015-01-28:
If someone is digging into this section of code, one idea that I have always wanted to see (but don't have the time) is to have an option that uses a better option than the average normal. I would love to see an option that used the median angle, so that with a more advanced approach to loosevar, would end up much closer mathematically to "properly downsampling" higher resolution surface. In the very least any thoughts on where it might fit would be interesting to hear.

Eric89GXL, 2015-01-28:
I don't know enough about surface downsampling to know what the ideal solution is. If there are improvements to be made along the lines of using the median angle, I wouldn't expect them to be too difficult to implement (a few lines) or put in the API (kwarg). But it would be worth looking to see 1) how big the changes actually were, and 2) what affect this had on localization, if any, before actually putting them in `master`.

Eric89GXL, 2015-12-12:
@agramfort do you still want changes for this?

agramfort, 2015-12-12:
I could live with this for 10 months... closing

closed 2015-12-12 by agramfort