mne-tools/mne-python #1767: edf reader fails

aestrivex, 2015-01-29:
I have some iEEG data in .edf format. mne_edf2fiff works just fine. read_edf_file() is able to read the metadata, but when it tries to read the iEEG data it crashes.

```
In [3]: ra = mne.io.edf.read_raw_edf('raw.edf', preload=True)
Extracting edf Parameters from raw.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading raw data from /local_mount/space/truffles/2/users/rlaplant/ecog/raw.edf...
Reading 0 ... 3781782  =      0.000 ...  7567.997 secs...
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)

/local_mount/space/truffles/1/users/rlaplant/.canopyrc/User/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/io/edf/edf.py in _read_segment(self, start, stop, sel, verbose, projector)
    276                     print int(sfreq), n_chan, blocks
    277                     data = data.reshape((int(sfreq), n_chan, blocks),
--> 278                                         order='F')
    279                     for i in range(blocks):
    280                         start_pt = int(sfreq * i)
```

It is trying to reshape a 232352736x1 matrix into a 499x129x7568 matrix.

It don't even know where to begin debugging this -- it seems like it is not figuring out correct information in the header? Maybe there is some encoding issue?

@dgwakeman suggested to tag @teonlamont

teonlamont, 2015-01-29:
First, just to check, have you updated to the most recent master? There were some bugs that were resolved over the past couple of days. 
Do you have a file you could share to help troubleshoot?

The reshaping is done because edf data are written in segments in time (records or blocks) and it will ultimately need to be in a channel by time.

aestrivex, 2015-01-29:
Yes this is on a semi-recent dev version.

I will try it again soon and update.

teonlamont, 2015-02-03:
@aestrivex I found a solution to the problem with pr #1771.

closed 2015-02-03 by teonlamont