mne-tools/mne-python #1770: FIX? Channel selection

yousrabk, 2015-02-03:
The **mne.selection.read_selection** doesn't take into account the **raw/epochs.info['ch_names']**, so we still have problems about channels defined with space or without e.g. 'MEG 1623' (in selection) and 'MEG1623' (raw file).

What is the best way to fix this? And actually, why can't we directly pick types by using e.g. *selection='temporal'* instead of list of channels. That way we have the info.

Thoughts?

agramfort, 2015-02-03:
add a note about the fact than include and selection are somehow redundant.

dengemann, 2015-02-03:
> add a note about the fact than include and selection are somehow redundant.

include is a list of channels, selection is a category / group, right?

@yousrabk look at mne.utils._clean_names

yousrabk, 2015-02-03:
@dengemann actually both include and selection are list of channels.

agramfort, 2015-02-03:
yes but we need info to call clean_names and read_selection does not allow
this... hence do it in pick_types

dengemann, 2015-02-03:
Seems like you would like to discuss our pick_types API. If we make changes we should go deep and change it in a conistent way that will last.

dengemann, 2015-02-03:
> The only sane default would be for pick_types to not select any channels by default. Any channel type you want, and channel type you don't want (bads for example), you explicitly specify!

that's really uncomfortable and leads to redundant code. Why not just return the data channels?

wmvanvliet, 2015-02-03:
Say I want to plot the EOG channels. Why can't `pick_types(raw.info, eog=True)` to actually return the EOG channels? Why does it have to assume I don't only want the EOG, but also other stuff?

dengemann, 2015-02-03:
>  Why does it have to assume I don't only want the EOG, but also other stuff?

because most of the time you want other stuff :)

wmvanvliet, 2015-02-03:
You can go that route. But then actually the behaviour as it is now is preferable.
If the default is to enable all data channel types that are present in the data, simply selecting the EOG channels would become:

    pick_types(raw.info, meg=False, meg_ref=False, eeg=False, seeg=False, ecg=False, emg=False, misc=False, eog=True)

that is, if I don't want to make any prior assumptions on what types of data might be available or not and really only want EOG, always. The current behaviour of just having to de-select meg and ref_meg is a lot less worse :)

dengemann, 2015-02-03:
@wmvanvliet 

```Python

def pick_types(info, meg=True, eeg=False, stim=False, eog=False, ecg=False,
               emg=False, ref_meg='auto', misc=False, resp=False, chpi=False,
               exci=False, ias=False, syst=False, seeg=False,
               include=[], exclude='bads', selection=None):
```

eeg is off by default.

agramfort, 2015-02-03:
how do you remember that type can be chpi if you don't see it in signature? also I don't buy the argument of purity or elegance. Breaking backward compat for purity reasons is evil. I think we have more pressing matters to address.

dengemann, 2015-02-03:
We should at least change the default, eeg=False doesn't really make any
sense.

agramfort, 2015-02-03:
I am too lazy to handle the deprecation pain... for little gain.

dengemann, 2015-02-03:
You're getting old ;-)

wmvanvliet, 2015-02-03:
@agramfort +1 for backwards compatibility > purity.

agramfort, 2015-02-03:
>
> You're getting old ;-)
>

and wise :)

Eric89GXL, 2017-02-16:
I think @jaeilepp fixed this at some point, but feel free to reopen if I'm mistaken

closed 2017-02-16 by Eric89GXL