mne-tools/mne-python #1773: eliminate_stim_artifact for evoked

agramfort, 2015-02-04:
eliminate_stim_artifact only accepts raw objects. It makes more sense to remove the artifact on evoked.

eliminate_stim_artifact should accept evoked object.

any taker?

dlsrnsi, 2015-03-21:
I think its solution is that copy notch_filter method in .io.base.py to evoked.py.

agramfort, 2015-03-21:
what do you mean the solution?

dlsrnsi, 2015-03-21:
If you copy notch_filter method in .io.base.py to evoked.py, you can eliminate stim_artifact for evoked, I think.

agramfort, 2015-03-21:
? stim artifact is not corrected with notch_filter. See how it is done with
raw files.

dlsrnsi, 2015-03-21:
Oops.. Sorry for my mistake.. I have misunderstood about this issue..

agramfort, 2015-03-21:
issue should be simpler than dealing with notch filter

mainakjas, 2015-03-21:
It's `eliminate_stim_artifact` not eliminate stim_artifact :)

Take a look here: https://github.com/mne-tools/mne-python/search?utf8=%E2%9C%93&q=eliminate_stim_artifact

dlsrnsi, 2015-04-13:
@agramfort Now we can close this issue.

closed 2015-04-13 by dengemann