mne-tools/mne-python #1780: Time-frequency confusion

choldgraf, 2015-02-05:
Not sure if this is intentional or not, but I find that the time-frequency module is quite cluttered. It seems that many of the functions / classes have been added at different times, and with often overlapping functionality. Moreover, there are some inconsistencies in terms of functionality (e.g., there's a cwt_morlet function, but no cwt_multitaper to my knowledge).

My specific use-case is that I'm trying to create a TFR representation with multitapers, but I don't want to average over trials yet. There are a few reasons for me wanting to do so, but for now an example is that I'd like to calculate the envelope of an audio signal by summing across frequency bands.

However, all of the tfr_XXXX methods return an AverageTFR object (signals x freqs x times), whereas I want the full TFR representation (epochs x signals x freqs x times). It's not clear to me whether this is something that MNE supports, but seems like it could at least be a flag to the various tfr_XXXX functions (e.g., "average_tfr=True" by default).

Eric89GXL, 2015-02-05:
I think you can probably get what you want by directly iterating over epochs using something like:
```
tfrs = [cwt_morlet(epoch, ...) for epoch in epochs]
```
Have you tried this? It will have some added overhead over your proposal, but it might be small compared to the actual calculation time.

Eric89GXL, 2015-02-05:
... in any case I agree there is some inconsistency in the naming and such. +1 for a PR that cleans it up with proper deprecation.

agramfort, 2015-02-05:
to address your usecase you can use the fact that you can index epochs

[tfr_XXXX(epochs[k], ...) for k in range(len(epochs)]

should do the trick but I agree that we need to cleanup stuff. The new
tfr_XXX functions are already a big step forward compared to 1 year ago.

Eric89GXL, 2015-02-05:
@agramfort as suggested above you can also just iterate directly as `[tfr_XXX(epoch, ...) for epoch in epochs]`, right?

choldgraf, 2015-02-05:
I just tried iterating and it did work, which is really cool. I didn't realize you could index epochs :)

I agree that the TFR methods are quite nice, I assume those are the most recent additions. If anything else, it would be good to ensure that any PSD or TFR function allows you to pass arbitrary frequencies rather than fmin and fmax.

agramfort, 2015-02-05:
> I agree that the TFR methods are quite nice, I assume those are the most recent additions. If anything else, it would be good to ensure that any PSD or TFR function allows you to pass arbitrary frequencies rather than fmin and fmax.

why is that? the freqs are defined by Fourier bins. You want to downsample?

agramfort, 2015-02-05:
> @agramfort <https://github.com/agramfort> as suggested above you can also
> just iterate directly as [tfr_XXX(epoch, ...) for epoch in epochs], right?
>

this will iterate of epochs.get_data()

Eric89GXL, 2015-02-05:
The code doesn't actually call `get_data()`, it just goes through and gets the next non-bad epoch...

choldgraf, 2015-02-05:
> why is that? the freqs are defined by Fourier bins. You want to downsample?

as one example, in one case I wanted to take log-spaced bins to reduce computation time. This doesn't happen when defining frequencies with fmin/fmax, right?

agramfort, 2015-02-05:
yes yes It's meant to do the same as iterating over get_data()

agramfort, 2015-02-05:
>
> as one example, in one case I wanted to take log-spaced bins to reduce
> computation time. This doesn't happen when defining frequencies with
> fmin/fmax, right?
>
if you want log space bins I'd use morlet ie wavelets that expects an array
of freqs and n_cycles as input

choldgraf, 2015-02-05:
> if you want log space bins I'd use morlet ie wavelets that expects an array
> of freqs and n_cycles as input

Yup - that's what I've been using so far. I think that iterating epochs by indexing them should work for this case. I guess if you only wanted the PSD rather than full TFR, you could just average over > 0 timepoints too.

agramfort, 2015-02-05:
> Yup - that's what I've been using so far. I think that iterating epochs by indexing them should work for this case. I guess if you only wanted the PSD rather than full TFR, you could just average over > 0 timepoints too.

this would not be a valid PSD unless you use stockwell computed with
timepoints > 0

I'd recommend a proper PSD computation if you want this.

choldgraf, 2015-02-05:
I don't have much experience creating PSDs from a pre-existing time-frequency representation, so I'm deferring to you on that one :) I was trying to think of a case where you could calculate a PSD while supplying custom frequencies, but perhaps this is not a valid thing to do?

Eric89GXL, 2015-02-05:
With Fourier transforms it's frowned upon because you're not guaranteed orthogonality of the transformation basis vectors.

agramfort, 2015-02-05:
>
> I don't have much experience creating PSDs from a pre-existing
> time-frequency representation, so I'm deferring to you on that one :) I was
> trying to think of a case where you could calculate a PSD while supplying
> custom frequencies, but perhaps this is not a valid thing to do?
>
if you call is PSD then it's not valid. If you call it the sum of the
squared coefs of the morlet .... along the time axis it's ok :)

choldgraf, 2015-02-05:
@Eric89GXL but if you're calculating the TFR from the DPSS, aren't those orthogonal to one another? Either way, what you and @agramfort say makes sense! I need to brush up on my signal processing :P

Eric89GXL, 2015-02-05:
DPSS tapers (window functions) are orthogonal to one another, yes. DPSS is basically a fancy way of windowing signals for Fourier decomposition, which is a separate issue from the orthogonality of the Fourier bases.

choldgraf, 2015-02-05:
Makes sense, thanks for the clarification.

choldgraf, 2015-02-05:
PS sorry for de-railing this conversation with questions about TF decomposition. Regarding the original issue, I'm not sure what exactly should be changed, as all of the functions in time_frequency do seem to work, it's just that there are some differences in usage and such.

Eric89GXL, 2015-02-05:
Unify their naming / usage / APIs, of course :)

If you want to undertake the task, the easiest first step might be to list the set of function signatures that we have that you're interested in unifying, then propose the way you think they all should look so yourself and future users are happier with them. IIRC correcly the function signatures were at least fairly similar, so maybe it would just be renaming functions. That would be nice, since deprecating functions via renaming is an easy case.

choldgraf, 2015-02-06:
OK cool - I can try to focus on this a bit later. I want to make sure that the other PRs I opened up a few weeks back are ready to go before starting new things.

Eric89GXL, 2015-02-06:
Awesome.

Eric89GXL, 2015-04-22:
@choldgraf I assume you won't have time to work on this in the next day or two? I might have time to tackle it, or just move it to 0.10.

choldgraf, 2015-04-22:
I'll focus on making the topo plotting stuff finalized, since that's pretty close (I think, anyway). I am happy to iterate on this one with you guys and review code etc. Does that sound good?

Eric89GXL, 2015-04-22:
Yep, sounds good -- thanks for your help

choldgraf, 2015-04-22:
Of course - thanks for the gentle reminders ;)

Eric89GXL, 2015-04-22:
According to our docs, this is our current API:
```
compute_raw_psd(raw[, tmin, tmax, picks, ...])
compute_epochs_psd(epochs[, picks, fmin, ...])
cwt_morlet(X, sfreq, freqs[, use_fft, ...])
tfr_morlet(inst, freqs, n_cycles[, use_fft, ...])
tfr_multitaper(inst, freqs, n_cycles[, ...])
single_trial_power(data, sfreq, frequencies)
stft(x, wsize[, tstep, verbose])
istft(X[, tstep, Tx])
stftfreq(wsize[, sfreq])
tfr_stockwell(inst[, fmin, fmax, n_fft, ...])
```
I think the first step is to split these into whether they operate on `ndarrays` or Python objects as:
```
compute_raw_psd(raw[, tmin, tmax, picks, ...])
compute_epochs_psd(epochs[, picks, fmin, ...])
tfr_morlet(inst, freqs, n_cycles[, use_fft, ...])
tfr_multitaper(inst, freqs, n_cycles[, ...])
tfr_stockwell(inst[, fmin, fmax, n_fft, ...])

cwt_morlet(X, sfreq, freqs[, use_fft, ...])
single_trial_power(data, sfreq, frequencies)
stft(x, wsize[, tstep, verbose])
istft(X[, tstep, Tx])
stftfreq(wsize[, sfreq])
```
In this sense, the functions that operate on mne-python instances are fairly uniform. I could clean them up a bit, though. The `ndarray` ones could use some tweaking. Did you have anything in mind other than this @choldgraf?

choldgraf, 2015-04-22:
This sounds like a good start to me. It'd also be good to standardize the input arguments (e.g., sometimes it's called `X`, other times `x`, other times `data`. If there's a difference an this is intention, then cool, but if they all just mean "an ndarray" then it should be the same input value.

Regarding functionality, my main confusing when I first played around with this is probably similar to what most people will have. I just wanted to know which function was the most up-to-date one, and it wasn't clear from the docs. It seems like the `tfr_` functions are more recent, but it wasn't obvious. Also the naming conventions were a bit wonky. E.g., for the `tfr` functions, the general structure is `output_method-of-output`. For the `psd` functions, it's `verb_type-of-input_output`. Why not call them `psd_epochs` rather than `compute_epochs_psd`. Stuff like that would make it easier to sort out the functions for a newbie.

Another thought is that many of the `ndarray` functions are called by the `tfr` functions. It might be helpful to include this in the docstring (something like "see `cwt_morlet`" or something like this).

Thoughts?

agramfort, 2015-04-22:
I like the

from mne.time_frequency import psd_epochs, psd_raw

+1 for see also in docstrings too.

any other thought?

choldgraf, 2015-04-22:
I guess for me a general principle I've found useful is to assume that the
person will be using tab completion in an interactive session to find the
function they want. E.g., if I want to do a TFR analysis, I'll type
`mne.time_frequency.tfr` and then hit tab. In an ideal world, that would
thing bring up all possible TFR-style functions. I think of function naming
as a process of going from general to specific. This isn't always perfect
but it helps me remember what things do :)

Eric89GXL, 2015-04-22:
@agramfort do we need to deprecate `compute_raw_psd` and `compute_epochs_psd`, then? Have we made a release with them?

Also, perhaps we should start adding `..versionadded` signifiers to our functions so that I don't need to ask questions like this...

agramfort, 2015-04-23:
>
> @agramfort <https://github.com/agramfort> do we need to deprecate
> compute_raw_psd and compute_epochs_psd, then?
>
I guess so...

> Have we made a release with them?
>
not sure...

PR #2021 (2015-04-26): FIX: Refactor TF functions and many docstrings
Closes #1780.

@agramfort I think we might want to re-think the renaming, see below...
closed 2015-04-26 by agramfort