mne-tools/mne-python #1781: apply_function argument error

choldgraf, 2015-02-05:
The apply_function method of Raw objects currently accepts the following arguments:

self, fun, picks, dtype, n_jobs, verbose=None, *args, **kwargs

Unless I'm missing a trick on passing arguments to functions, this will throw an error unless you set verbosity level with a positional argument rather than a flag.

E.g., say you passed these arguments: (my_fun, [0], None, 1, verbose=None, arg1, arg2=my_arg)

You'd raise the error that non-keyword arguments have been given after a keyword argument.

If you use this input:

(my_fun, [0], None, 1, arg1, arg2=my_arg)

Then it will assume that arg1 is the positional argument for "verbose" and will only pass arg2 to the function.

The only way I got this to work was to set verbose to a positional argument, e.g.:

(my_fun, [0], None, 1,None, arg1, arg2=my_arg)

If this is the only way of doing it, then verbose should either have to be a positional argument, or the way that arguments are handled should be a bit different. E.g., doesn't pandas have an "args" keyword where you pass a tuple with the positional arguments to pass to the function?

Eric89GXL, 2015-02-05:
Maybe we should just do `(self, fun, picks. dtype, n_jobs, *args, **kwargs)`, keep `verbose` in the docstring, but then do `kwargs.pop('verbose', None)` in the code.

agramfort, 2015-02-05:
>
> Maybe we should just do (self, fun, picks. dtype, n_jobs, *args, **kwargs),
> keep verbose in the docstring, but then do kwargs.pop('verbose', None) in
> the code.
>

if this works +1

PR #1948 (2015-04-11): FIX: Fix apply_function order
Closes #1781.
closed 2015-04-11 by agramfort