mne-tools/mne-python #1787: Losing samples when using reference with read_raw_brainvision

jona-sassenhagen, 2015-02-09:
read_raw_brainvision loses samples when a reference is set.

```
>>>raw = read_raw_brainvision(vdhr_fname, preload=True, verbose=False, montage=montage_file)
>>>raw2 = read_raw_brainvision(vdhr_fname, preload=True, verbose=False, montage=montage_file, reference='A2')
>>>print('n_times without reference: ', raw.n_times, " and n_times with reference: ", raw2.n_times)
n_times without reference:  901750  and n_times with reference:  874424
```

The correct number is 901750, and the data has 32 channels (reference not included), so we are losing times/channels worth of data.
There is no A2 in the .vhdr of the files I am importing.

Should I upload the data and/or full script somewhere?

teonlamont, 2015-02-09:
I think this error is coming about because the datablock is based on the number of channels and the ref code is adding another channel which is changing the data structure. https://github.com/mne-tools/mne-python/blame/master/mne/io/brainvision/brainvision.py#L445 and https://github.com/mne-tools/mne-python/blame/master/mne/io/brainvision/brainvision.py#L180

cc: @christianbrodbeck

teonlamont, 2015-02-09:
found the culprit: https://github.com/mne-tools/mne-python/blame/master/mne/io/brainvision/brainvision.py#L99

the last sample was calculated based on the number of channels. did not change when reference was added.

teonlamont, 2015-02-09:
closing for pr https://github.com/mne-tools/mne-python/pull/1789.

teonlamont, 2015-02-09:
thanks @jona-sassenhagen for catching this bug!

closed 2015-02-09 by teonlamont