mne-tools/mne-python #1790: Deprecate Brainvision internal referencing

teonlamont, 2015-02-09:
Brainvision reader has a separate internal referencing method. This should be deprecated in favor of the new referencing method in  `mne.io.reference`.

agramfort, 2015-02-09:
any taker?

@wmvanvliet ?

wmvanvliet, 2015-02-10:
> Brainvision reader has a separate internal referencing method
It does? As far as I can see, it just adds an all-zeros channel, which is useful for re-referencing later. This re-referencing is then done with the methods in `mne.io.reference`.

agramfort, 2015-02-10:
do we have now 2 ways to do EEG referencing with BrainVision data?

teonlamont, 2015-02-10:
Could we add the adding the all-zero channels to the reference module? This would make a clean separation of reading module from the referencing one. Also it would make it consistent and easier to maintain.

wmvanvliet, 2015-02-10:
adding a channel (being zero or not) to a dataset should be implemented at a higher level. An operation that concatenates two datasets along the first (=channels) dimension.

Something vaguely like this:

    raw = mne.io.read_raw_brainvision(...)
    ref_channel = mne.RawArray(np.zeros((1, n_samples)),
                               mne.create_info(ch_names=['REF'], ch_types=['eeg']))
    raw = mne.concatenate_channels([raw, ref_channel])

Perhaps it already exists.

wmvanvliet, 2015-02-10:
Or another approach: the `set_eeg_reference` function could take an optional `prev_ref` parameter:

    reref, _ = mne.io.set_eeg_reference(raw, ['A1', 'A2'], prev_ref=['Fz'])

which would first add an all-zeros channel 'Fz' and then re-reference the entire thing to the mean of 'A1' and 'A2'.

agramfort, 2015-02-10:
the raw object should have a state that says the previous reference applied.

any array concat logic should be hidden from the user.

wmvanvliet, 2015-02-10:
> any array concat logic should be hidden from the user.

Ok. We could store `prev_eeg_reference` as a field in the info dict and make `set_eeg_reference` take this into account. I would still opt to additionally provide a `prev_ref` parameter to deal with the cases in which the info structure does not have the `prev_eeg_reference` field (older raws read from disk, RawArray's created by the user, etc.)

agramfort, 2015-02-10:
if you use RawArray it's your responsibility to define the eeg_ref in the
info. I'd like to keep the info simple and prefer not to address all crazy
usecases.

Eric89GXL, 2015-03-09:
Deprecation would be good to get in before 0.9 release if this seems doable.

teonlamont, 2015-03-14:
closing for #1858.

closed 2015-03-14 by teonlamont