mne-tools/mne-python #1792: Difficulties applying baseline to epochs object

choldgraf, 2015-02-10:
Right now it is possible to supply a baseline time window during the creation of an Epochs object, but not to designate the baseline mode (e.g., "zscore"). There is a "rescale" function in the baseline module, but this operates on numpy arrays. Since there isn't an "apply_function" method for Epochs objects, it seems like the options for doing a custom baseline on an Epochs object is to apply "rescale" to the raw data of the Epochs object, then re-create an EpochsArray object with all the same metadata. E.g.:

```python
def baseline_epoch(epochs, baseline, mode):
    baselined_epochs = mne.baseline.rescale(epochs.get_data(), epochs.times, baseline, mode, )
    epochs = mne.EpochsArray(baselined_epochs, epochs.info, epochs.events, epochs.tmin,
                             event_id=epochs.event_id, reject=epochs.reject, etc...)
```

This seems like a really clunky way of doing this. It seems like a couple of things might make this simpler:

1. Add an "apply_baseline" method to the Epochs object that would let you baseline whenever you'd like.
2. Add a "baseline_epochs" function to the baseline module that would accept an epochs object rather than an array and return an Epochs object.
3. More generally, add an "epochs_like" function that would basically allow you to create a new Epochs object of the exact same data shape as a pre-existing one, which has all of the same metadata, etc. The only difference would be the values of the data.
4. More generally, add an "apply_function" method to the Epochs object that would let you apply an arbitrary function to each epoch x signal item.
5. Make it possible to designate the baselining "mode" in the Epoch creation.

Thoughts on this? Feel free to correct me if I'm missing a helper function somewhere.

Eric89GXL, 2015-02-10:
`1.` is fine by me, but I'd amend it to say that, for safety reasons (at least to start), baselining can only be done once. Thus if you want to use `apply_baseline`, then you need to instantiate with `baseline=None` in the constructor.

`4.` seems like it might actually be more general, so I think I prefer that one most of all.

`5` I don't like because we probably already have too many constructor args. `2.` and `3.` seem too cumbersome.

choldgraf, 2015-02-10:
Yeah I agree - I've been constructing Epochs objects with no baseline, then just applying the rescale function to the raw data (generally a zscore on the whole trial followed by a mean baseline, this can help with visualizing epoch traces where some signals have a much higher variance)

agramfort, 2015-02-11:
I agree with eric. What should be the required parameters of the applied
function? 2d data and times?

Eric89GXL, 2015-02-11:
Sounds reasonable. I was going to say that we should require epochs to be preloaded, but it actually just occurred to me that we don't actually need to require that. We could set up a queue like `self._load_functions = []` in the constructor, and then `apply_function` appends the function call to that queue. But maybe that would be for a subsequent PR once someone hit a use case where preloading was too painful :)

agramfort, 2015-02-11:
maybe an occasion to simplify Epochs (spaghetti?) code? :)

wmvanvliet, 2015-02-11:
+1 for simplifying Epochs code.
`self._load_functions` sounds really great. Application of the SSP could be one of the functions in that list.

choldgraf, 2015-02-11:
+1 for simplifying Epochs code, but another +1 for doing that in a
subsequent PR :)

wmvanvliet, 2015-02-12:
such a scheme would also allow for EEG referencing without requiring the epochs to be pre-loaded.

Definitely a subsequent PR.

Getting back to the use case at hand, while implementing the FASTER routines (#1777) I also came across the need to redo the baseline on the epochs after some filtering.

agramfort, 2015-02-12:
I'll review any PR :)

Eric89GXL, 2015-12-12:
We have `apply_baseline` now, closing

closed 2015-12-12 by Eric89GXL

choldgraf, 2015-12-12:
I didn't know that!