mne-tools/mne-python #1793: Create slider for bem

teonlamont, 2015-02-11:
Hey @mainakjas, is it possible to make the bem html sliders if you want to script your own report? currently, it is returning only an array of bem images in my html file.

mainakjas, 2015-02-11:
mmm ... the function that makes the slider is private: https://github.com/mne-tools/mne-python/blob/master/mne/report.py#L1392. So, no

dengemann, 2015-02-11:
I'm having the same use case + question:)

mainakjas, 2015-02-11:
Maybe we should make the function public then :) How about:

```python
html = report.render(subject, subjects_dir, decim, n_jobs)
report.add_html_to_section(html, caption, section)
```

That way, we have maximum flexibility so that any arbitrary html text can be added

teonlamont, 2015-02-11:
+1. that way we could later extend it so that if we wanted add some text to
the section, we could write a function that writes the html object.

-teon

Teon Brooks, Pre-hD
NSF Fellow, Chateaubriand Fellow
PhD Candidate
Department of Psychology
New York University

teonlamont, 2015-03-14:
closing for #1859.

closed 2015-03-14 by teonlamont