mne-tools/mne-python #1807: read_layout cannot be imported

sstober, 2015-02-18:
Hello,

I'm trying to use the read_layout() function in mne/channels/layout.py

from mne.channels import read_layout
results in "ImportError: cannot import name read_layout"

from mne.channels.layout import read_layout
results in "ImportError: No module named layout"

How can I use that function?

Cheers,
Sebastian

sstober, 2015-02-18:
Looks like this was caused by a strange version conflict with an old version of mne that was not properly uninstalled through pip. Sorry for the confusion! It's working now as expected and I'm closing the issue.

closed 2015-02-18 by sstober

Eric89GXL, 2015-02-18:
Glad you were able to sort this out. Could you share how you found the problem? If at some point we add a FAQ it might be nice to include debugging steps people have taken.

I'm guessing you did something like:
```Python
>>> import mne
>>> mne
<module 'mne' from '...'>
>>> mne.__version__
'0.9.dev'
```
and found a value you didn't expect for one of those?

sstober, 2015-02-19:
At some point, a function complained about an unknown argument that was obviously present in its source code. Digging deeper into this issue, I found that my code was referring to an older mne version which I had installed (and uninstalled) through pip while I thought I was working with the most recent version I cloned from github. The lines you proposed would have revealed this. Eventually, I had to manually remove the old version as pip thought it was already uninstalled.

agramfort, 2015-02-20:
@sstober looking forward your deeplearning progress using mne ;)

let us know how it goes.