mne-tools/mne-python #1808: ENH: include rERP / rERF

wmvanvliet, 2015-02-18:
A few weeks ago, Nathaniel J. Smith published a paper that proposes a pretty straightforward enhancement to the way we compute evokeds. I'm leaving this here as a reminder for myself to implement and evaluate this.

The papers are interesting and extremely accessible to read:

http://vorpus.org/papers/smith-kutas-2015-rerps-1.pdf
http://vorpus.org/papers/smith-kutas-2015-rerps-2-with-si.pdf

The author even supplies a reference python implementation for the method.

agramfort, 2015-02-18:
looks similar in spirit to:

http://martinos.org/mne/stable/auto_examples/stats/plot_sensor_regression.html#example-stats-plot-sensor-regression-py

wmvanvliet, 2015-02-18:
It's a continuation of Hauk's work. Looks like the meat of the implementation is already present in `mne.stats.linear_regression`, awesome!

agramfort, 2015-02-18:
Looks like we have a pr on the way with an example? ;)

jona-sassenhagen, 2015-02-19:
I have implemented this, specifically the "overlap correction" variant described by Smith that is a major improvement over the method used by Hauk et al, in EEGLAB myself and had been planning on translating it to MNE anyways, since I think it's an exceptionally interesting method. But I haven't started yet.
The big difference is that the "overlap correction" method models the whole EEG, not just selected epochs, which really changes things when you have overlapping epochs.

[Here is the unfinished Python implementation](https://github.com/rerpy/rerpy) by NJ Smith.

The principle is actually very simple, and in many ways simpler than even condition averaging.

agramfort, 2015-02-19:
it's sad to see this code unfinished...

anyone wants to mentor for such a project as a GSOC?

please add it to the list.

teonlamont, 2015-04-10:
hey, is there still time to have this project with GSOC? I wouldn't mind mentoring someone on this

jona-sassenhagen, 2015-04-10:
I'd actually like to do this once at least 2 of my currently open PRs are done.

Eric89GXL, 2015-04-10:
Proposals were due a few weeks ago, unfortunately

teonlamont, 2015-04-10:
ok, maybe this doesn't require too much work. @jona-sassenhagen sounds good

-teon

jona-sassenhagen, 2015-04-10:
You can still mentor me Teon :)

I'll start working on it once I've gotten corrmap merged.

jona-sassenhagen, 2015-07-08:
@wmvanvliet can you add the code sprint label to this?

jona-sassenhagen, 2015-07-17:
Can probably close for #2304 now.

wmvanvliet, 2015-07-17:
close in favor of #2304

closed 2015-07-17 by wmvanvliet