mne-tools/mne-python #1811: ENH/FIX: croping will leave out samples if .times vector includes some rounding errors, etc.

dengemann, 2015-02-20:
imagine you have 1 time sample -0.004000000000000000000036 that you compare against tmin -0.004

the croping code will use naive comparison against tmin and tmax:

instead of

```Python
tmask = (self.times >= tmin) & (self.times <= tmax)```
```
use 

```Python
tmask = (((self.times > tmin) | np.isclose(self.times, tmin)) &
              ((self.times < tmax) | np.isclose(self.times, tmax)))
```

We should probably have utils function that generalizes and unifies time sample selection.

we need to do an exclusive less / greater in conjunction with np.isclose to avoid leaving out samples 

FYI this has been an issue when comparing results between pipelines.

I'm sure several parts of our code are affected by this issue.

cc @Eric89GXL @agramfort @fraimondo

agramfort, 2015-02-20:
sounds legit to me.

we should list the places that need this change

Eric89GXL, 2015-03-09:
+1 for doing this as a fix for 0.9. I might have time this week to tackle this if it seems worthwhile.

PR #1855 (2015-03-17): FIX: Use cleaner time comparison
Closes #1811. Note that this also hopefully fixes a few off-by-one inclusion errors that existed in previous slices.

Ready for review/merge from my end.
closed 2015-03-17 by dengemann