mne-tools/mne-python #1814: faster resampling using scikits.samplerate

sstober, 2015-02-20:
Hi,

I've noticed that resample is very slow. (I tried to resample ~1h of EEG with 69 channels from 512Hz to 100Hz on 4 cores and gave up after 1 hour of waiting.)

Here's an alternative and much faster way using scikits.samplerate (through librosa - but that's just for convenience) that takes less than 1 minute one 1 core for the same data:
https://gist.github.com/sstober/260ab1da3025b0216675

Maybe that would be an option for mne...

Cheers,
Sebastian

Eric89GXL, 2015-02-20:
Sounds great, feel free to open a PR. We will need to verify the extent to which scipy's FFT-based algorithm matches. BTW have you tried the CUDA filtering? I'm curious which one would be faster.

Eric89GXL, 2015-02-20:
Oh wait, nevermind. The licencse is GPL, so we cannot use it unfortunately. Same problem with FFTW. (The main slowness of our algorithms comes from the FFT / IFFT slowness of fftpack.)

agramfort, 2015-02-20:
yes but a factor 60s??? I would understand a factor 4 but not more.

Eric89GXL, 2015-02-20:
I think MATLAB avoids FFTs / IFFTs altogether by using some polyphasic routines I don't fully understand. It's likely that the `libsamplerate` routines do something similar. Since they're used primarily for audio AFAIK, it's entirely possible that some of the algorithms take minor shortcuts that minimally affect the sound of the files, too.

agramfort, 2015-02-20:
do we want to use such an approach for EEG?

Eric89GXL, 2015-02-20:
It would be great, but someone needs to implement it in a BSD-friendly way. I could probably figure out how to do it given the right papers/textbooks (probably textbooks) but it would definitely take me some time. It would probably go best in `scipy`, especially since then I could use Cython once I got a pure-python version working. But in any case, it probably won't be easy.

Incidentally, the idea of using polyphasic routines eventually is why I formatted our `resample` routines to all use `up` and `down` (ideally integers) instead of a simple `ratio`, since I'm pretty sure those routines require it.

agramfort, 2015-02-20:
it's worth asking on scipy dev. Maybe someone can share some code.

sstober, 2015-02-20:
Sorry, I did not consider licensing issues. 

Also, I have to admit I do not know what tricks they are using "under the hood". I've done some visual tests plotting a short EEG segment at high resolution and I cannot tell which approach gives the better result.

I've tried CUDA, but it complained about not enough memory. (I'm using a retina MBP and the GPU can only handle small problem aside from running the GUI)

jona-sassenhagen, 2015-03-06:
The EEGLAB implementation is very fast, even the alternative one that doesn't use the signal processing toolbox. I don't understand the code, but here it is: https://code.google.com/p/eeglab/source/browse/eeglab/functions/popfunc/pop_resample.m

I assume the alternative implementation is open.

Eric89GXL, 2015-03-06:
Open yes, BSD-compatible licensing (GPL) no. So that would be a problem in translating the code.

Looking at the code, it first tries to do `upfirdn`-based resampling (standard MATLAB sig proc toolbox resampling), but then falls back to 1D spline interpolation. I've never actually seen that used for processing data. I imagine it introduces some artifacts. Unless someone knows that spline interpolation is safe, we shouldn't go that route. upfirdn should be a better algorithm if I can find time to push it into `scipy`.

Eric89GXL, 2015-08-27:
closing as something we probably can't do due to licensing issues

closed 2015-08-27 by Eric89GXL