mne-tools/mne-python #1819: mne.viz.ica._plot_ica_overlay_evoked() broken (no red color)

sstober, 2015-02-22:
Hello,

running this example
martinos.org/mne/stable/auto_examples/preprocessing/plot_ica_from_raw.html
the call of
ica.plot_overlay(eog_evoked, exclude=eog_inds) 
(2nd last command)
produces only black lines. 

It looks like this could be fixed by using 
show=False
in line 457 of mne.viz.ica._plot_ica_overlay_evoked()

Cheers,
Sebastian

agramfort, 2015-02-22:
the figures do not render as in the web page for you?

dengemann, 2015-02-22:
I'm surprised, this code is in use every day. Can you say more about how
you use it and on which system with wich python distribution?
Maybe a reproducible example?

sstober, 2015-02-22:
You can use the plot_ica_from_raw example. These are the problematic figures:
[image]
[image]

I'm on Mac OS 10.8.5 using system Python 2.7.8. mne is the current master version through git, matplotlib (1.4.2) is build with homebrew using the backend MaxOSX ("version unknown"). Same result if I force "Qt4Agg" as backend.

agramfort, 2015-02-22:
hum... so you get this just when running the untouched example?

if so can you try with anaconda?

dengemann, 2015-02-22:
That's crazy. We cannot reproduce this. I suspect it has to do with the
matplotlib build. You should use the 'osx' with mavericks.

sstober, 2015-02-22:
I'm afraid there's no way I'm upgrading to Mavericks as it does not properly support some software I still depend on. I'd also rather not tinker with my otherwise perfectly functioning Python setup as it took me some time to get Theano and Pylearn2 working nicely together with Cuda support on my GPU.
Guess I'll use my custom plotting function (with show=False) for now.

dengemann, 2015-02-22:
Sorry my mistake I thought you were using Mavaricks. It would nevertheless be good to explore what's going on. I fear the problem is in matplotlib not in the mne-python code. I cannot really help a lot because I cannot reproduce it.

agramfort, 2015-02-22:
sounds fair.

Sorry if we cannot help more.

sstober, 2015-02-22:
If you cannot reproduce it, I'd suspect mpl as well. I'll see if I can find out more about this. But for now I got more pressing stuff to attend to and I can live with my workaround.

closed 2015-03-09 by dengemann

nbara, 2018-08-24:
FWIW I'm also experimenting this on macOS 10.12 (I do not see a red trace)

However it works again when I do this:

```python
import matplotlib.pyplot as plt
...
ica.plot_overlay(eog_evoked, exclude=eog_inds, show=False)
plt.show()
```