mne-tools/mne-python #1821: ENH: anatomy + meg script in python

dengemann, 2015-02-22:
We should have a funciton that runs all these steps we usually do via the shell scripts `run_anatomy_tutorial.sh` and `run_meg_tutorial.sh`. Ideally it would also reimplement the bem scripts in python to avoid the weaknesses with respect to path and environment variables. Ideall it would just take the subejcts dir and run everything for eachs subject.

cc @Eric89GXL @agramfort @LauriParkkonen @mainakjas

agramfort, 2015-02-28:
@lorenzo-desantis might be able to help

kambysese, 2015-03-11:
@dengemann what do these shell scripts do? Where can I find them?I am working on a script that runs the <i>anatomy</i> routines for a given subject that I just recently sent to @Eric89GXL for comment--that is how I ended up here. It's a [WIP](https://gist.github.com/kambysese/4bdaec63fb6e12b84150) and a rather crude first pass at pythonizing a boat load of shell commands. LMK if it can be of any use, unless @agramfort or @lorenzo-desantis have something more elegant in the works, I can work on improving this.

agramfort, 2015-03-11:
please share the gist so we can see what you started

kambysese, 2015-03-11:
Here it is in my dropbox folder (https://www.dropbox.com/s/nsp6gocbbxufhoc/FSrecon.py?dl=0)
btw what happens when you follow the above WIP hyperlink to the gist? it should take you to a public gist, no? :confused:

agramfort, 2015-03-11:
thx for sharing

Eric89GXL, 2017-03-21:
I think we have all of the functionality in Python now. @dengemann do you still see a need for a unified script / command?

larsoner, 2019-08-20:
This is now out of scope here and is the goal of `mne-study-template` and similar efforts

closed 2019-08-20 by larsoner