mne-tools/mne-python #1823: IO of SSS info is broken after filtering with mne_process_raw

agramfort, 2015-02-22:
To reproduce, take the somato data, filter with:

mne_process_raw --raw sef_raw_sss.fif --highpass 1 --lowpass 40 --save test_raw.fif --projoff --decim 2 --filtersize 16384

then try this:

```
import mne

info = mne.io.read_info('test_raw.fif')
mne.io.write_info('tmp_info.fif', info)
info = mne.io.read_info('tmp_info.fif')
```

it should break. In otherwords the MNE-C code breaks the Python IO of SSS metadata :( :(

@Eric89GXL bad weather in seattle?

agramfort, 2015-02-22:
@mshamalainen there is a bug in the writing of sparse matrices in the C code. The sparse matrix in maxfilter proc history gets messed up on writing back to disk.

can you have a look?

BrainFPV, 2015-03-03:
It was working in the 0.8 release.. bisecting now.

mluessi, 2015-03-03:
It was working in the 0.8 release, running bisect now.

dengemann, 2015-03-03:
> It was working in the 0.8 release, running bisect now.

I'm not sure I'm following, this functionality was recently added.

Eric89GXL, 2015-03-03:
0.8 didn't read the SSS info, now we do. @agramfort is implying the problem is with the C code, not the Python code.

mluessi, 2015-03-03:
Oh, I see.. makes sense. So the error only happens after running mne\_process\_raw, correct?

mshamalainen, 2015-03-03:
How did @agramfort <https://github.com/agramfort> come to that conclusion? What exactly happens as a manifestation of the problem? The C code seems to work flawlessly reading and writing sparse matrices.

> 
> 0.8 didn't read the SSS info, now we do. @agramfort <https://github.com/agramfort> is implying the problem is with the C code, not the Python code.

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.

mshamalainen, 2015-03-03:
It does not happen to me even then.

> 
> Oh, I see.. makes sense. So the error only happens after running mne_process_raw, correct?

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.

dengemann, 2015-03-03:
@mluessi with @agramfort we processed data and realized that the fiff files containing SSS info are not readable anymore after the raw files have been processed using mne_process_raw + decimation. To be precise, the problem occured when evoked or epochs objects were created based on the processed raw files and saved to disk using python.

dengemann, 2015-03-03:
We were able to reproduce this using different SSSed datasets.

mshamalainen, 2015-03-03:
I believe now that this is the case but it is difficult to see how the C code would be broken in this respect. Let me investigate a little bit further.

> 
> We were able to reproduce this using different SSSed datasets.

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.

agramfort, 2015-03-03:
keep us posted @mshamalainen

mshamalainen, 2015-03-03:
There was, indeed, an error in the C code writing of sparse matrices. Fixed now. It was originally tested on my PowerPC MacBook and never came up because the C code never tried to actually use the sparse matrices it wrote. :)

dengemann, 2015-03-03:
@mshamalainen you rock! Thanks!!

mluessi, 2015-03-03:
I think it would make sense to also catch the error in the Python code and byte swapping in Python if it occurs. So it will work with raw files written by all versions of MNE. I volunteer to implement it.

agramfort, 2015-03-03:
PR more than welcome :)

mshamalainen, 2015-03-03:
This fix will be in the nightly build of MNE tomorrow.

agramfort, 2015-03-03:
mne-c.git clone up to date

mluessi, 2015-03-03:
There is a git clone of MNE-C? Where?

agramfort, 2015-03-03:
https://github.com/agramfort/mne-c

it's private though

agramfort, 2015-03-03:
btw if you want to take over the migration to git it would be great...

On Tue, Mar 3, 2015 at 10:26 PM, Alexandre Gramfort
<alexandre.gramfort@m4x.org> wrote:
> https://github.com/agramfort/mne-c
>
> it's private though

closed 2015-03-04 by dengemann