mne-tools/mne-python #1833: Filtering if info["highpass"] is None

jona-sassenhagen, 2015-03-04:
I have Brainvision files with a very truncated header. The BV import filter knows how to deal with these - it sets info["highpass"] and info["lowpass"] to None. This however causes an error when attempting to filter the data:

```
Filtering ...
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-11-0dd729552dea> in <module>()
     16 print("Filtering ...")
---> 17 raw_f.filter(1, 50, n_jobs=1, method="iir")

/home/FB/fb05/sassenha/tools/mne-python/mne/io/base.py in filter(self, l_freq, h_freq, picks, filter_length, l_trans_bandwidth, h_trans_bandwidth, n_jobs, method, iir_params, verbose)

/home/FB/fb05/sassenha/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    532         # set it back if we get an exception
    533         try:
--> 534             return function(*args, **kwargs)
    535         finally:
    536             set_log_level(old_level)

/home/FB/fb05/sassenha/tools/mne-python/mne/io/base.py in filter(self, l_freq, h_freq, picks, filter_length, l_trans_bandwidth, h_trans_bandwidth, n_jobs, method, iir_params, verbose)
    366             # and it's not a band-stop filter
    367             if h_freq is not None and (l_freq is None or l_freq < h_freq) and \
--> 368                     h_freq < self.info['lowpass']:
    369                 self.info['lowpass'] = h_freq
    370             if l_freq is not None and (h_freq is None or l_freq < h_freq) and \

TypeError: unorderable types: float() < NoneType()
```

The easiest solution is for the BV filter to set info["highpass”] to 0 and info["lowpass"] to the sampling rate if no online filtering info is found (no idea what the other importers do). Alternatively, ```raw.filter``` can be adapted. 
(I can probably do a PR for either of these options.)

agramfort, 2015-03-04:
+1 for PR with filter less strict just logging the absence of l_freq and
h_freq

jona-sassenhagen, 2015-03-05:
Close for #1834

PR #1834 (2015-03-05): Protect filter against info["{high|low}pass"] None
Proposal for dealing with filtering data where info["highpass"] or info["lowpass"] == None. Currently, the filter compares these fields to a float, which throws an error. Now, instead of checking ```(h_freq < info["lowpass"])```, we check ```(info["lowpass"] is None or h_freq < info["lowpass"])```.

Resolves #1833
closed 2015-03-05 by jona-sassenhagen