mne-tools/mne-python #1836: run make linkcheck in doc

agramfort, 2015-03-08:
```
cd doc
make linkcheck
```

and you'll see some red lines about broken links etc...

Full log on my machine:

```
writing output... [  0%] _templates/class
writing output... [  0%] _templates/function
writing output... [  0%] advanced_setup
(line   46) ok        https://developer.nvidia.com/cuda-downloads
(line   50) ok        http://wiki.tiker.net/PyCuda/Installation/
(line   48) ok        http://mathema.tician.de/software/pycuda
(line   52) ok        https://github.com/lebedov/scikits.cuda
writing output... [  0%] auto_examples/connectivity/plot_cwt_sensor_connectivity
writing output... [  1%] auto_examples/connectivity/plot_mne_inverse_coherence_epochs
writing output... [  1%] auto_examples/connectivity/plot_mne_inverse_connectivity_spectrum
writing output... [  1%] auto_examples/connectivity/plot_mne_inverse_label_connectivity
writing output... [  1%] auto_examples/connectivity/plot_mne_inverse_psi_visual
writing output... [  2%] auto_examples/connectivity/plot_sensor_connectivity
writing output... [  2%] auto_examples/datasets/plot_megsim_data
writing output... [  2%] auto_examples/datasets/plot_megsim_data_single_trial
writing output... [  2%] auto_examples/datasets/plot_spm_faces_dataset
writing output... [  3%] auto_examples/debug_proj
writing output... [  3%] auto_examples/decoding/plot_decoding_csp_eeg
(line   13) ok        http://en.wikipedia.org/wiki/Common_spatial_pattern
writing output... [  3%] auto_examples/decoding/plot_decoding_csp_space
writing output... [  3%] auto_examples/decoding/plot_decoding_sensors
writing output... [  4%] auto_examples/decoding/plot_decoding_spatio_temporal_source
writing output... [  4%] auto_examples/decoding/plot_decoding_time_generalization
writing output... [  4%] auto_examples/decoding/plot_ems_filtering
writing output... [  4%] auto_examples/export/plot_epochs_as_data_frame
(line   89) ok        http://pandas.pydata.org/pandas-docs/stable/
writing output... [  5%] auto_examples/export/plot_epochs_to_nitime
writing output... [  5%] auto_examples/export/plot_evoked_to_nitime
writing output... [  5%] auto_examples/export/plot_raw_to_nitime
writing output... [  5%] auto_examples/index
(line   41) -local-   ./plot_bem_contour_mri.html
(line   41) -local-   ./plot_ssp_projs_topomaps.html
(line   41) -local-   ./plot_coregistration_transform.html
(line   41) -local-   ./plot_read_noise_covariance_matrix.html
(line   41) -local-   ./read_events.html
(line   41) -local-   ./plot_read_evoked.html
(line   41) -local-   ./plot_topography.html
(line   41) -local-   ./plot_meg_eeg_fields_3d.html
(line   41) -local-   ./plot_decimate_head_surface.html
(line   41) -local-   ./plot_estimate_covariance_matrix_raw.html
(line   41) -local-   ./plot_extract_events_from_raw.html
(line   41) -local-   ./plot_ssp_projs_sensitivity_map.html
(line   41) -local-   ./plot_read_bem_surfaces.html
(line   41) -local-   ./plot_read_and_write_raw_data.html
(line   41) -local-   ./plot_evoked_topomap.html
(line   41) -local-   ./plot_read_epochs.html
(line   41) -local-   ./plot_shift_evoked.html
(line   41) -local-   ./plot_interpolate_bad_channels.html
(line   41) -local-   ./plot_topo_channel_epochs_image.html
(line   41) -local-   ./plot_estimate_covariance_matrix_baseline.html
(line   41) -local-   ./debug_proj.html
(line   41) -local-   ./plot_evoked_topomap_delayed_ssp.html
(line   41) -local-   ./plot_topo_customized.html
(line   41) -local-   ./plot_channel_epochs_image.html
(line   41) -local-   ./plot_rereference_eeg.html
(line   41) -local-   ./plot_read_forward.html
(line   41) -local-   ./plot_topo_compare_conditions.html
(line   41) -local-   ./plot_evoked_whitening.html
(line   41) -local-   ./plot_from_raw_to_multiple_epochs_to_evoked.html
(line   41) -local-   ./plot_make_forward.html
[... 723 lines cut ...]
(line  702) broken    https://github.com/christianmbrodbeck - HTTP Error 404: Not Found
(line  704) broken    https://github.com/christianmbrodbeck - HTTP Error 404: Not Found
(line  706) broken    https://github.com/christianmbrodbeck - HTTP Error 404: Not Found
(line  706) redirect  http://faculty.washington.edu/larsoner/ - with Found to http://staff.washington.edu/larsoner/
(line  708) broken    https://github.com/christianmbrodbeck - HTTP Error 404: Not Found
(line  708) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  712) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  714) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  759) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  767) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  769) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  781) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  783) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  810) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  812) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  816) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  822) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  824) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  828) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  853) broken    http://www.nmr.mgh.harvard.edu/martinos/people/showPerson.php?people_id=1600 - HTTP Error 403: Forbidden
(line  253) ok        https://github.com/dongqunxi
(line  336) ok        https://github.com/chdinh
(line  230) ok        http://www.researchgate.net/profile/Lukas_Breuer
(line  494) ok        http://simonster.com
```

Eric89GXL, 2015-03-08:
+1 for fixing this, plus adding some cleaning of the `make linkcheck` to only display lines that are actually useful (maybe just containing "broken").

agramfort, 2015-03-08:
@TanayGahlot could be an easy fix for you

TanayGahlot, 2015-03-08:
Cool, I will take it up

PR #1854 (2015-03-13): FIX: Fix links
Closes #1836.

Ready for review/merge @agramfort.
closed 2015-03-13 by agramfort