mne-tools/mne-python #1837: [ENH] release MNE-Python 0.9

dengemann, 2015-03-09:
@christianbrodbeck @teonlamont @joewalter @wmvanvliet @agramfort @mluessi @Eric89GXL @mainakjas @rgoj .......

We should make a release, it's overdue. Can we please tackle the open issues and PRs that are close to merge? Any reviewring or coding contrib is welcome.
Please list what you want for MNE-Python 0.9.

I think for met it's only #1629, #1795 and #1637.

Eric89GXL, 2015-03-09:
Instead of listing, I made a 0.9 milestone and will tag the appropriate issues/PRs.

dengemann, 2015-03-09:
Thanks Eric!

teonlamont, 2015-03-09:
also, let's try to close any open issue that has been resolved or no longer applicable.

Eric89GXL, 2015-04-10:
Closing this one so it's not in the list of things to finish for 0.9

closed 2015-04-10 by Eric89GXL