mne-tools/mne-python #1839: Q: shift time axis

kingjr, 2015-03-10:
Is there a clean way of shifting the time axis on epoched and evoked data or woudld the following be sufficient?
<code>
time_shift = 0.100  # 100 ms time shift
epochs.times += time_shift
epochs.tmin += time_shift
epochs.tmax += time_shift
</code>

Thanks!

agramfort, 2015-03-10:
shift the events it's cleaner and guaranteed to work

kingjr, 2015-03-10:
ok. (Although I think I a function handling this a posteriori would be popular... http://fieldtrip.fcdonders.nl/reference/ft_redefinetrial)

closed 2015-03-10 by kingjr

agramfort, 2015-03-10:
hum ok... it's more complete than just a time shift.

you could open an issue with a detailed description of what you would like
to be able to do to mimic fieldtrip, just typing code the way you would
like to use it. A PR will come later

kingjr, 2015-03-10:
ok.