mne-tools/mne-python #1840: infomax discrepancies relative to EEGLAB binica/runica

pntfx4, 2015-03-10:
@dengemann @cle1109 @kazemakase

In comparing the mne-python implementation of infomax to that done using EEGLAB binica/runica I tweaked the code 9 lines below 'if not wts_blowup:' immediately before '#anneal learning rate' to mimic what is done by binica/runica: 

    print('step %d - lrate %7f, wchange %8.8f, angledelta %4.1f deg' % (step, l_rate, change, angledelta))

Running it through matlab - then using the bang command to execute the python code; the python implementation is substantially faster. However, in my test file binica/runica take over 360 steps whereas using the mne-python implementation a solution is resolved in only 66 steps. The placement of the output calls in python to show the steps is similar to that of runica, and the parameters were specified to be the same as used by runica so it seems odd that I am getting a solution with substantially fewer steps.

kazemakase, 2015-03-12:
cc @jdammers

dengemann, 2015-03-12:
> I tweaked the code 9 lines below 'if not wts_blowup:' immediately before '#anneal learning rate' to mimic what is done by binica/runica:

Can you show the diff? Does this create the expected behavior? I remember with @fraimondo we also observed different iteration numbers in the past when comparing to his cuda implementation of infomax.
This would be good to investigate and understand.

jdammers, 2015-03-12:
@pntfx4  Can you confirm starting with the same learning rate. Did you observe angledelta being <60 ?
Would it be possible to post the output (print statements) of both versions?

pntfx4, 2015-03-12:
Ok, I edited line 133 of the infomax function to output the same parameters as runica:

        print('Input data size [%d, %d] = %d channels, %d frames' % (n_features, n_samples, n_features, n_samples))
        if extended is True:
            print('Finding %d ICA components using extended ICA.' % n_features)
            print('Kurtosis will be calculated initially every %d blocks using %d data points.' % (ext_blocks, kurt_size))
        else:
            print('Finding %d ICA components.' % n_samples)
        print('Initial learning rate will be %7f, block size %d.' % (l_rate, block))
        print('Learning rate will be multiplied by %3f whenever angledelta >= %3f deg.' % (anneal_step, anneal_deg))
        print('Stopping weight change %7f.' % w_change)
        print('Training will end when w_change < %7f or after %d steps.' % (w_change, max_iter))

Doing so, highlights two differences in the settings.
1) the anneal_step defaults to 0.9 in both functions, but in runica when running the extended option the default switches to 0.98.
2) the block size calculation differs:
mne-python: 

     block = int(math.floor(math.sqrt(n_samples / 3.0)))

runica:

    DEFAULT_BLOCK = ceil(min(5*log(frames),0.3*frames));

pntfx4, 2015-03-12:
This is the runica output:

Input data size [61,150000] = 61 channels, 150000 frames/n
Finding 61 ICA components using extended ICA.
Kurtosis will be calculated initially every 1 blocks using 6000 data points.
Decomposing 40 frames per ICA weight ((3721)^2 = 150000 weights, Initial learning rate will be 0.001, block size 60.
Learning rate will be multiplied by 0.98 whenever angledelta >= 60 deg.
More than 32 channels: default stopping weight change 1E-7
Training will end when wchange < 1e-07 or after 512 steps.
Online bias adjustment will be used.
Removing mean of each channel ...
Final training data range: -1902.15 to 1455.62
Computing the sphering matrix...
Starting weights are the identity matrix ...
Sphering the data ...
Beginning ICA training ... first training step may be slow ...
step 1 - lrate 0.001000, wchange 51.87717819, angledelta  0.0 deg
step 2 - lrate 0.001000, wchange 6.78059495, angledelta  0.0 deg
step 3 - lrate 0.001000, wchange 6.62994524, angledelta 91.6 deg
step 4 - lrate 0.000980, wchange 6.76851707, angledelta 116.4 deg
Lowering learning rate to 0.00086436 and starting again.
step 1 - lrate 0.000864, wchange 48.07021380, angledelta  0.0 deg
step 2 - lrate 0.000864, wchange 5.03102550, angledelta  0.0 deg
step 3 - lrate 0.000864, wchange 4.70670946, angledelta 86.5 deg
step 4 - lrate 0.000847, wchange 4.70303340, angledelta 107.2 deg
step 5 - lrate 0.000830, wchange 4.83686135, angledelta 115.5 deg
step 6 - lrate 0.000814, wchange 4.41025064, angledelta 113.8 deg
step 7 - lrate 0.000797, wchange 4.20544925, angledelta 114.6 deg
step 8 - lrate 0.000781, wchange 3.79352194, angledelta 116.3 deg
step 9 - lrate 0.000766, wchange 3.29437510, angledelta 114.6 deg
step 10 - lrate 0.000750, wchange 4.13096367, angledelta 120.7 deg
step 11 - lrate 0.000735, wchange 4.24569316, angledelta 118.0 deg
step 12 - lrate 0.000721, wchange 3.80864237, angledelta 123.0 deg
step 13 - lrate 0.000706, wchange 3.10394296, angledelta 111.9 deg
step 14 - lrate 0.000692, wchange 3.48887428, angledelta 111.4 deg
step 15 - lrate 0.000678, wchange 3.07136473, angledelta 112.4 deg
step 16 - lrate 0.000665, wchange 3.82720046, angledelta 114.6 deg
step 17 - lrate 0.000651, wchange 3.91644500, angledelta 124.2 deg
step 18 - lrate 0.000638, wchange 2.94403313, angledelta 117.9 deg
step 19 - lrate 0.000626, wchange 2.65697811, angledelta 120.8 deg
step 20 - lrate 0.000613, wchange 3.61984809, angledelta 115.0 deg
step 21 - lrate 0.000601, wchange 2.71158004, angledelta 119.7 deg
step 22 - lrate 0.000589, wchange 2.72453968, angledelta 107.6 deg
step 23 - lrate 0.000577, wchange 3.13800269, angledelta 122.0 deg
step 24 - lrate 0.000566, wchange 2.45615503, angledelta 112.4 deg
step 25 - lrate 0.000554, wchange 3.08991606, angledelta 116.9 deg
step 26 - lrate 0.000543, wchange 2.95618024, angledelta 114.5 deg
step 27 - lrate 0.000532, wchange 2.63534665, angledelta 114.5 deg
step 28 - lrate 0.000522, wchange 2.84640210, angledelta 107.8 deg
step 29 - lrate 0.000511, wchange 2.76210166, angledelta 119.1 deg
step 30 - lrate 0.000501, wchange 3.40618432, angledelta 111.2 deg
step 31 - lrate 0.000491, wchange 2.81526927, angledelta 128.4 deg
step 32 - lrate 0.000481, wchange 2.51387294, angledelta 118.2 deg
step 33 - lrate 0.000471, wchange 2.09780257, angledelta 118.4 deg
step 34 - lrate 0.000462, wchange 1.94891423, angledelta 107.6 deg
step 35 - lrate 0.000453, wchange 1.91033641, angledelta 116.8 deg
step 36 - lrate 0.000444, wchange 2.03234172, angledelta 111.3 deg
step 37 - lrate 0.000435, wchange 1.78332959, angledelta 108.7 deg
step 38 - lrate 0.000426, wchange 1.90388531, angledelta 116.5 deg
step 39 - lrate 0.000418, wchange 2.01017855, angledelta 119.5 deg
step 40 - lrate 0.000409, wchange 1.43567589, angledelta 135.2 deg
step 41 - lrate 0.000401, wchange 1.46589041, angledelta 98.4 deg
step 42 - lrate 0.000393, wchange 2.49209817, angledelta 110.8 deg
step 43 - lrate 0.000385, wchange 1.88599239, angledelta 112.8 deg
step 44 - lrate 0.000378, wchange 0.88658375, angledelta 112.3 deg
step 45 - lrate 0.000370, wchange 2.53210105, angledelta 103.1 deg
step 46 - lrate 0.000363, wchange 1.77267538, angledelta 126.6 deg
step 47 - lrate 0.000355, wchange 0.90735573, angledelta 116.5 deg
step 48 - lrate 0.000348, wchange 2.11158700, angledelta 109.9 deg
step 49 - lrate 0.000341, wchange 1.89019191, angledelta 126.6 deg
[... 439 lines cut ...]
step 489 - lrate 0.000001, wchange 0.00009117, angledelta  4.3 deg
step 490 - lrate 0.000001, wchange 0.00005883, angledelta  5.0 deg
step 491 - lrate 0.000001, wchange 0.00003883, angledelta  6.0 deg
step 492 - lrate 0.000001, wchange 0.00002606, angledelta  7.8 deg
step 493 - lrate 0.000001, wchange 0.00001766, angledelta  9.8 deg
step 494 - lrate 0.000001, wchange 0.00001215, angledelta 10.6 deg
step 495 - lrate 0.000001, wchange 0.00000865, angledelta 14.1 deg
step 496 - lrate 0.000001, wchange 0.00000611, angledelta 15.9 deg
step 497 - lrate 0.000001, wchange 0.00000437, angledelta 18.6 deg
step 498 - lrate 0.000001, wchange 0.00000321, angledelta 22.1 deg
step 499 - lrate 0.000001, wchange 0.00073591, angledelta 177.4 deg
step 500 - lrate 0.000001, wchange 0.00083703, angledelta  2.0 deg
step 501 - lrate 0.000001, wchange 0.00081585, angledelta  2.0 deg
step 502 - lrate 0.000001, wchange 0.00001629, angledelta 168.3 deg
step 503 - lrate 0.000001, wchange 0.00028008, angledelta 12.0 deg
step 504 - lrate 0.000001, wchange 0.00017051, angledelta 11.8 deg
[... cut ...]
step 506 - lrate 0.000001, wchange 0.00007007, angledelta 12.5 deg
step 507 - lrate 0.000001, wchange 0.00004636, angledelta 12.4 deg
step 508 - lrate 0.000001, wchange 0.00003130, angledelta 13.2 deg
step 509 - lrate 0.000001, wchange 0.00002139, angledelta 14.0 deg
step 510 - lrate 0.000001, wchange 0.00001486, angledelta 14.4 deg
step 511 - lrate 0.000001, wchange 0.00001031, angledelta 16.1 deg
step 512 - lrate 0.000001, wchange 0.00000747, angledelta 18.3 deg
Sorting components in descending order of mean projected variance ...

pntfx4, 2015-03-12:
mne-python infomax output:

Input data size [61, 150000] = 61 channels, 150000 frames
Finding 61 ICA components using extended ICA.
Kurtosis will be calculated initially every 1 blocks using 6000 data points.
Initial learning rate will be 0.001000, block size 60.
Learning rate will be multiplied by 0.980000 whenever angledelta >= 60.000000 deg.
Stopping weight change 0.000001.
Training will end when w_change < 0.000001 or after 512 steps.
step 1 - lrate 0.001000, wchange 52.67882020, angledelta  0.0 deg
step 2 - lrate 0.001000, wchange 5.82420635, angledelta 94.1 deg
step 3 - lrate 0.000980, wchange 5.72881183, angledelta 115.1 deg
... lowering learning rate to 0.00086436 ... re-starting...
step 1 - lrate 0.000864, wchange 51.90372539, angledelta  0.0 deg
step 2 - lrate 0.000864, wchange 5.46734905, angledelta 92.4 deg
step 3 - lrate 0.000847, wchange 4.45154226, angledelta 114.4 deg
step 4 - lrate 0.000830, wchange 4.44782608, angledelta 110.6 deg
... lowering learning rate to 0.000732176 ... re-starting...
step 1 - lrate 0.000732, wchange 49.96906430, angledelta  0.0 deg
step 2 - lrate 0.000732, wchange 4.84623992, angledelta 93.6 deg
step 3 - lrate 0.000718, wchange 3.58630188, angledelta 109.7 deg
step 4 - lrate 0.000703, wchange 3.63633851, angledelta 109.2 deg
step 5 - lrate 0.000689, wchange 4.40026550, angledelta 120.4 deg
step 6 - lrate 0.000675, wchange 3.90777510, angledelta 116.5 deg
step 7 - lrate 0.000662, wchange 3.41433799, angledelta 114.4 deg
step 8 - lrate 0.000649, wchange 4.11262973, angledelta 117.1 deg
step 9 - lrate 0.000636, wchange 3.15260687, angledelta 115.2 deg
step 10 - lrate 0.000623, wchange 3.52011160, angledelta 119.6 deg
step 11 - lrate 0.000610, wchange 3.27832438, angledelta 118.5 deg
step 12 - lrate 0.000598, wchange 3.01569066, angledelta 116.3 deg
step 13 - lrate 0.000586, wchange 3.08790244, angledelta 114.5 deg
step 14 - lrate 0.000575, wchange 2.45671510, angledelta 117.1 deg
step 15 - lrate 0.000563, wchange 3.31814273, angledelta 121.8 deg
step 16 - lrate 0.000552, wchange 2.58697311, angledelta 116.9 deg
step 17 - lrate 0.000541, wchange 2.88306321, angledelta 110.3 deg
step 18 - lrate 0.000530, wchange 2.69661108, angledelta 119.7 deg
step 19 - lrate 0.000519, wchange 2.41973216, angledelta 113.6 deg
step 20 - lrate 0.000509, wchange 3.29485055, angledelta 114.7 deg
step 21 - lrate 0.000499, wchange 3.02822856, angledelta 116.5 deg
step 22 - lrate 0.000489, wchange 1.91307112, angledelta 112.2 deg
step 23 - lrate 0.000479, wchange 2.18280605, angledelta 112.0 deg
step 24 - lrate 0.000469, wchange 2.09818053, angledelta 105.0 deg
step 25 - lrate 0.000460, wchange 1.65898879, angledelta 111.3 deg
step 26 - lrate 0.000451, wchange 2.46808761, angledelta 117.7 deg
step 27 - lrate 0.000442, wchange 1.85883299, angledelta 111.6 deg
step 28 - lrate 0.000433, wchange 1.94494009, angledelta 111.9 deg
step 29 - lrate 0.000424, wchange 1.70627225, angledelta 111.0 deg
step 30 - lrate 0.000416, wchange 1.78494531, angledelta 107.7 deg
step 31 - lrate 0.000408, wchange 1.52451051, angledelta 113.3 deg
step 32 - lrate 0.000399, wchange 1.86706587, angledelta 132.5 deg
step 33 - lrate 0.000391, wchange 1.53576515, angledelta 120.2 deg
step 34 - lrate 0.000384, wchange 2.59388389, angledelta 120.6 deg
step 35 - lrate 0.000376, wchange 2.66926004, angledelta 130.0 deg
step 36 - lrate 0.000368, wchange 1.73147242, angledelta 140.6 deg
step 37 - lrate 0.000361, wchange 2.17779053, angledelta 124.4 deg
step 38 - lrate 0.000354, wchange 2.16539739, angledelta 135.4 deg
step 39 - lrate 0.000347, wchange 1.61789590, angledelta 132.7 deg
step 40 - lrate 0.000340, wchange 1.42127941, angledelta 123.2 deg
step 41 - lrate 0.000333, wchange 1.53736290, angledelta 106.8 deg
step 42 - lrate 0.000326, wchange 1.53911619, angledelta 125.9 deg
step 43 - lrate 0.000320, wchange 0.62474446, angledelta 91.3 deg
step 44 - lrate 0.000313, wchange 1.02558473, angledelta 116.7 deg
step 45 - lrate 0.000307, wchange 1.12462050, angledelta 109.7 deg
step 46 - lrate 0.000301, wchange 0.85574646, angledelta 112.9 deg
step 47 - lrate 0.000295, wchange 0.71132999, angledelta 102.7 deg
step 48 - lrate 0.000289, wchange 1.78734250, angledelta 114.7 deg
step 49 - lrate 0.000283, wchange 0.97204779, angledelta 130.9 deg
step 50 - lrate 0.000278, wchange 1.34258042, angledelta 112.0 deg
step 51 - lrate 0.000272, wchange 0.91203323, angledelta 120.1 deg
step 52 - lrate 0.000267, wchange 1.36973858, angledelta 130.9 deg
[... 255 lines cut ...]
step 308 - lrate 0.000002, wchange 0.00000598, angledelta 108.6 deg
step 309 - lrate 0.000002, wchange 0.00000654, angledelta 109.6 deg
step 310 - lrate 0.000002, wchange 0.00000424, angledelta 93.2 deg
step 311 - lrate 0.000001, wchange 0.00000520, angledelta 99.8 deg
step 312 - lrate 0.000001, wchange 0.00000412, angledelta 99.5 deg
step 313 - lrate 0.000001, wchange 0.00000400, angledelta 99.4 deg
step 314 - lrate 0.000001, wchange 0.00000405, angledelta 93.4 deg
step 315 - lrate 0.000001, wchange 0.00000415, angledelta 97.2 deg
step 316 - lrate 0.000001, wchange 0.00000323, angledelta 93.2 deg
step 317 - lrate 0.000001, wchange 0.00000407, angledelta 106.3 deg
step 318 - lrate 0.000001, wchange 0.00000392, angledelta 88.5 deg
step 319 - lrate 0.000001, wchange 0.00000311, angledelta 103.5 deg
step 320 - lrate 0.000001, wchange 0.00000238, angledelta 94.1 deg
step 321 - lrate 0.000001, wchange 0.00000190, angledelta 94.5 deg
[... cut ...]
step 325 - lrate 0.000001, wchange 0.00000209, angledelta 93.6 deg
step 326 - lrate 0.000001, wchange 0.00000223, angledelta 98.9 deg
step 327 - lrate 0.000001, wchange 0.00000131, angledelta 101.1 deg
step 328 - lrate 0.000001, wchange 0.00000144, angledelta 103.4 deg
step 329 - lrate 0.000001, wchange 0.00000120, angledelta 89.9 deg
step 330 - lrate 0.000001, wchange 0.00000143, angledelta 94.3 deg
step 331 - lrate 0.000001, wchange 0.00000126, angledelta 94.0 deg
step 332 - lrate 0.000001, wchange 0.00000093, angledelta 94.5 deg

Eric89GXL, 2015-03-12:
So what do you think is the best course of action?

In any case, I think it's worth switching your `print` statement over to `logger.debug` so this is easier to debug in the future. Feel free to open a PR.

agramfort, 2015-03-12:
we should try to make sure both implementation are identical by default
eventually exposing expert parameters if required.

+1 for PR

cle1109, 2015-03-13:
Even if the settings are identical, the differences are still significant IMHO. What's more, I think EEGLAB tries to automatically set the number of sub-Gaussian sources for extended Infomax, but I don't think that this is an option in the MNE implementation.

In addition, maybe we should really use default Infomax as a default - first, this is what EEGLAB does, and second, there are less parameters to set.

jdammers, 2015-03-13:
In principle, I would agree that it would be nice if both versions (EEGlab and MNE-python) produce the same output, although to me the output of MNE looks better. It seems that with runica you are trapped within a local minimum (see angledelta and wchange jumping up and down) and the only reason why the algorithm stops is that it reaches the number of maximum iterations.  

The learning rate and the annealing factor are the most critical values in both Infomax (extended and standard). If you follow the history changes of the original runcia the default values (including block size) have also changes and are determined heuristically. It is also recommended to adjust these parameters depending on whether you use MEG or EEG... as MEG typically has a lot more channels.

In both cases I find the default learning rate to high as it will also be reduced by the algo itself (see output). The question is if we want to use 1) the same defaults for MEG and EEG (I think we shouldn't) and 2) if we even in EEG encounter non-optimal defaults; shall we still go for these defaults?

cle1109, 2015-03-13:
@jdammers if the defaults in EEGLAB are not sensible, we should not force ourselves to use these settings. Also, if you have good reasons to set different defaults for EEG and MEG, I would be in favor of this change.

Regardless of the specific steps produced by both algorithms, I think a comparison between resolved sources would be helpful. In general, these sources should be similar in both algorithms (and if not, we should show why there are differences - maybe EEGLAB gets stuck in local minima sometimes and therefore the resolved sources are not as good as in the MNE implementation).

jdammers, 2015-03-13:
@cle1109 agree and finally the output of the sources is of more importance.
Also note, the Default changes (we introduced) were motivated by the fact that modern MEG systems have 250 channels or more while about 20 years ago (when the algo was written) most EEG labs used 32 channels or less. We also encountered that the default learning rate of 0.001 was a bit too high for MEG.

pntfx4, 2015-03-13:
@jdammers @cle1109 I also agree that the defaults do not necessarily have to mirror EEGLAB. In fact a selling point might be that more appropriate defaults are used and perhaps a tweak is included to prevent getting stuck in local minima. As an outsider to this area, my focus is more on being able to utilize this as a platform independent replacement for binica. So any further improvements are icing on the cake. That being said, having the option to have the outputs be the same as runica/binica would likely ease most users transition (even if the default is to suppress the outputs). 
Along those lines another difference between the implementations is that runica/binica also sort the sources in descending order of mean projected variance. There may be good reason to not do this, but if not it would make the sources easier to compare.

I apologize if this is rudimentary, but conceptually, is there a reason that the block size calculation varies as a function of the number of data points rather than utilizing either a fixed size or having it vary based upon the sampling rate? I ask as I have noticed considerable variability in what that size is (anywhere from 50 to 300).

jdammers, 2015-03-14:
@pntfx4 First of all I would not recommend fixing the block size. 
We changed the block size estimation because most of us are primarily interest in having a decent block size for MEG (using a lot more channels compared to EEG). The runica version underestimates the size of each block as it was designed for EEG and a few channels only. If we have a sampling rate of 1000Hz and use 1 and 5 minutes of data; the difference in block size using runica would be 8 samples only (56 and 64) ; while in MNE-Python we will get 141 and 316 for the block sizes.
Agree in either version the "number of channels" is not included; neither the sampling rate which would be - if included - a more proper estimation. 
As Infomax is about pdf matching (joint pdf should be uniform) and if you have a lot more channels you also need more statistical power,  but I would not expect dramatically changes as long as your block size has a decent size (in other words; is large enough).
But making the block size too large is also not recommended. As we do temporal ICA we also need enough statistical power along time. So there will also be a compromise between block size and nblocks and of course the fact that we expect the data being stationary.

pntfx4, 2015-03-14:
@jdammers  Thank you! I did notice that binica has a similar block size to mne-python in looking at the outputs. I guess I was looking at the block size relative to how "meaningful" the training is if it was only considering 50 points (which at 1000hz is only 50ms if the data were to be continuous). So from that perspective, the overall number of points would seem to be less relevant. But it would seem then that the goal is more about having the training block size be "representative."

In looking at how the block size impacts the consistency of the P3 ERP component following removal of the eyeblink component across repeated ica runs (6) there does seem to be an optimal window where the variability in the ICA solutions impacts the stimulus-locked activity relatively little. Below are zoomed in screenshots of the FZ electrode (which manifests with the most variability, the same data file was run each time - 61 channels, 180880 points, 1000 hz sampling rate)

Block size 50:
[image]

Block size 150:
[image]

Block size 250:
[image]

Block size 300:
[image]

Block size 500:
[image]

Block size 1000:
[image]

Eric89GXL, 2015-04-22:
Unless someone here has time to do this in the next day or so, I'm going to move this to be dealt with in the 0.10 instead of 0.9 release

closed 2015-09-09 by agramfort