mne-tools/mne-python #1841: EHN: contrast default if comment is None

kingjr, 2015-03-10:
Doing a contrast between two conditions with 'event_id = None' leads to an error:
```python
condition1 = np.where(epochs.events[:,2] == 1)
condition2 = np.where(epochs.events[:,2] == 2)
evoked1 = epochs[condition1].average()
evoked2 = epochs[condition2].average()
contrast = evoked1 - evoked2
```
generates:

```python
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-36-a09bf137c719> in <module>()
----> 1 evoked_1 - evoked_2

~/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/evoked.pyc in __sub__(self, evoked)
    760         this_evoked.data *= -1.
    761         out = combine_evoked([self, this_evoked])
--> 762         out.comment = self.comment + " - " + this_evoked.comment
    763         return out
    764 

TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

I know this can be corrected manually by chaning the evoked.comment:

```
evoked1.comment = 'condition 1'
evoked2.comment = 'condition 2'
```

But I think it'd be easier to handle this automatically. What do you think?

Eric89GXL, 2015-03-10:
Yeah, this should be handled automatically, adding something like `this_comment = 'unknown if this_evoked.comment is None else this_evoked.comment` and a similar line for `other_comment` should do the trick. Feel free to open a PR. If you don't have time, let me know and I can do it.

kingjr, 2015-03-10:
I'll try to do it, it's a good exercize

dengemann, 2015-03-10:
Sounds good to me!

closed 2015-03-12 by Eric89GXL