mne-tools/mne-python #1843: Issue: redefine epoch time

kingjr, 2015-03-10:
It would be great to have a function that redefines or select the timing of epochs data. Following the fieldtrip function ft_redefinetrial (http://fieldtrip.fcdonders.nl/reference/ft_redefinetrial), I see two possible types of scenarios:
1. shift the time axis: 
``` epochs.times += 0.100 ```
2. troncate time axis:
```epochs.redefinetrial(tmin=-0.100, tmax=0.200)```

What do you think?

dengemann, 2015-03-10:
Sounds good. For the second use case we already have .crop -- I'll pay a beer to the one who puts some care in the crop method to make it consistent across mne containers, Raw, Epochs, etc

teonlamont, 2015-04-12:
I think the first use case will be accessible with #1952.

Eric89GXL, 2015-12-12:
This should now be possible, closing

closed 2015-12-12 by Eric89GXL