mne-tools/mne-python #1847: import mne from ipython gives runtime warning

flrgsr, 2015-03-12:
hi, 

Following [the getting started guide](http://martinos.org/mne/stable/getting_started.html) I just freshly installed mne via *pip install mne*.

However if I type *import mne* I get this:

```
/usr/local/lib/python2.7/site-packages/sklearn/utils/__init__.py:10: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from .murmurhash import murmurhash3_32
/usr/local/lib/python2.7/site-packages/sklearn/utils/__init__.py:10: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from .murmurhash import murmurhash3_32
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/__init__.py:3: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from ._min_spanning_tree import minimum_spanning_tree
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/__init__.py:3: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._min_spanning_tree import minimum_spanning_tree
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/_graph_validation.py:5: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from ._graph_tools import csgraph_to_dense, csgraph_from_dense,\
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/_graph_validation.py:5: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._graph_tools import csgraph_to_dense, csgraph_from_dense,\
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/__init__.py:4: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from ._traversal import connected_components
/usr/local/lib/python2.7/site-packages/sklearn/utils/sparsetools/__init__.py:4: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._traversal import connected_components
/usr/local/lib/python2.7/site-packages/sklearn/utils/extmath.py:20: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from ._logistic_sigmoid import _log_logistic_sigmoid
/usr/local/lib/python2.7/site-packages/sklearn/utils/extmath.py:20: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._logistic_sigmoid import _log_logistic_sigmoid
/usr/local/lib/python2.7/site-packages/sklearn/utils/extmath.py:22: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility
  from .sparsefuncs_fast import csr_row_norms
/usr/local/lib/python2.7/site-packages/sklearn/utils/extmath.py:22: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from .sparsefuncs_fast import csr_row_norms
```

How critical is this (as it is just a warning)? And what to do?

Cheers

agramfort, 2015-03-12:
yes it's just warnings but I don't know why

note that if you plan to contribute you should setup the current
development version.

Eric89GXL, 2015-03-17:
I have seen this in WinPython, or when I have custom-built the numpy/scipy stack but not packages that depend on it like `sklearn`. Try updating your versions and the warnings could go away. In any case, I don't think this is an mne-python specific issue, so I'm going to close this issue.

closed 2015-03-17 by Eric89GXL

cle1109, 2015-03-25:
I get similar warning messages when I `import mne`:

    /usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/importlib/_bootstrap.py:321: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
      return f(*args, **kwds)
    /usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/importlib/_bootstrap.py:321: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
      return f(*args, **kwds)
    /usr/local/Cellar/python3/3.4.3/Frameworks/Python.framework/Versions/3.4/lib/python3.4/importlib/_bootstrap.py:321: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
      return f(*args, **kwds) 

I found a similar bug with scikit-learn which has been resolved - could this be a related issue?
(https://github.com/scipy/scipy/issues/3863, https://github.com/scikit-learn/scikit-learn/issues/3537)

agramfort, 2015-03-25:
how was the pb solved with scipy+sklearn?

cle1109, 2015-03-26:
I think this was solved by https://github.com/scikit-learn/scikit-learn/commit/69af355dc257af - @ogrisel might know more about it.

agramfort, 2015-03-26:
can you see if it fixes the pb for you? I don't have the problem myself so
hard to test

cle1109, 2015-03-26:
Any idea where to look for these lines in MNE?

agramfort, 2015-03-26:
sorry no.

ogrisel, 2015-03-26:
I don't think https://github.com/scikit-learn/scikit-learn/commit/69af355dc257af is intended to "solve" such warnings in scikit-learn.

I would rather think that scipy was built against a different version of numpy than the one that is currently installed.

Maybe you can try to update the version of numpy and scipy you are using.

ogrisel, 2015-03-26:
But before launching the upgrades can you please give us the output of `pip freeze` on your host so that we can try to reproduce the issue?

cle1109, 2015-03-26:
alabaster==0.7.3
    Babel==1.3
    certifi==14.5.14
    docutils==0.12
    gnureadline==6.3.3
    ipython==3.0.0
    jdcal==1.0
    Jinja2==2.7.3
    jsonschema==2.4.0
    MarkupSafe==0.23
    matplotlib==1.4.3
    mistune==0.5.1
    mne==0.8.6
    nose==1.3.4
    numpy==1.9.2
    openpyxl==2.2.0
    pandas==0.16.0
    pep8==1.6.2
    ptyprocess==0.4
    pyflakes==0.8.1
    Pygments==2.0.2
    pyparsing==2.0.3
    python-dateutil==2.4.1
    pytz==2015.2
    pyzmq==14.5.0
    rope-py3k==0.9.4.post1
    rpy2==2.5.6
    scikit-learn==0.15.2
    scipy==0.15.1
    six==1.9.0
    snowballstemmer==1.2.0
    Sphinx==1.3.1
    sphinx-rtd-theme==0.1.7
    spyder==2.3.4
    statsmodels==0.6.1
    terminado==0.5
    tornado==4.1

I guess I have all the latest versions.

ogrisel, 2015-03-26:
Indeed this is weird. I don't recall observing those warnings with the same versions of those projects on my OSX laptop.

cle1109, 2015-03-26:
I think these warnings were always there since I first used MNE (I just ignored them). My setup is pretty standard, I used a brewed Python 3 and installed all Python packages with pip.

cle1109, 2015-03-26:
I can reproduce these warnings on another (clean) Mac OS X Yosemite:
- Install [Homebrew](http://brew.sh/)
- `brew install python3`
- `pip3 install numpy scipy nose mne`
- `python3 -c "import mne"`

cle1109, 2015-07-09:
Importing `mne` does not raise these warnings anymore, but something even weirder is happening now:

    import mne
    import pandas as pd

The `pandas` import now triggers these `numpy` warning messages. However, no warnings are produced if I reverse the import order:

    import pandas as pd
    import mne

Also, this problem is now specific to `pandas` - there are no warning messages if I import e.g. `numpy` or `scipy` after `mne`. Can anyone reproduce this? I'm on Mac OS X 10.10.4, Python 3.4.3 (via Homebrew), `mne` 0.9, `pandas` 0.16.2, `numpy` 1.9.2, `scipy` 0.15.1.

cle1109, 2015-07-09:
And here is another use case which triggers these warnings:

    import mne
    import scipy.stats

agramfort, 2015-07-09:
I cannot reproduce.

Can somebody?

cle1109, 2015-07-09:
I just reproduced it on a different Mac. Same setup though, I'm using a brewed Python 3.4.3 and pip installed all Python packages.

agramfort, 2015-07-09:
I cannot reproduce with anaconda.

try looking for things like:

warnings.simplefilter('default')

and commenting it to see if it helps.

Eric89GXL, 2015-07-09:
I've gotten these before when mixing which versions of numpy vs other libs (scipy, pandas, scikit-learn) I've had installed, usually it means they were not all compiled on the same system. Using `brew` and `pip` I would expect to hit these sorts of issues. They are related to the compilation of other libs, and mne-python just causes them to appear.

cle1109, 2015-07-09:
@Eric89GXL yeah, this sounds reasonable - does this mean that CPython installed via brew was compiled using different lib versions than pip packages? Because the only thing I install via brew is Python; I installed all packages (including numpy and mne) via pip.

What could I do to get rid of these messages? Surely this cannot be the expected behavior, but I don't even know where to report this problem.

Eric89GXL, 2015-07-09:
I've had the best experiences using Anaconda. I think most other folks here have had similar experiences.