mne-tools/mne-python #1849: BUG:

kingjr, 2015-03-12:
I seem to be encountering a bug similar to #1823 #1835 (although it was ok a few days ago). Specifically, I can read and transform an sss.fif file, save/write it, but when I want to read it again I get the following message:
```
Reading subject02_as/stim-subject02_as-epo.fif ...
---------------------------------------------------------------------------
IOError                                   Traceback (most recent call last)
<ipython-input-1-5ca5f765985f> in <module>()
      6 fname = 'subject02_as/stim-subject02_as-epo.fif'
----> 7 epochs = mne.read_epochs(data_path+fname)

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/epochs.pyc in read_epochs(fname, proj, add_eeg_ref, verbose)

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/utils.pyc in verbose(function, *args, **kwargs)
    535         finally:
    536             set_log_level(old_level)
--> 537     return function(*args, **kwargs)
    538 
    539 

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/epochs.pyc in read_epochs(fname, proj, add_eeg_ref, verbose)
   1901 
   1902     logger.info('Reading %s ...' % fname)
-> 1903     fid, tree, _ = fiff_open(fname)
   1904 
   1905     #   Read the measurement info

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/io/open.pyc in fiff_open(fname, preload, verbose)

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/utils.pyc in verbose(function, *args, **kwargs)
    535         finally:
    536             set_log_level(old_level)
--> 537     return function(*args, **kwargs)
    538 
    539 

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/io/open.pyc in fiff_open(fname, preload, verbose)
     95         while tag.next >= 0:
     96             pos = fid.tell()
---> 97             tag = read_tag_info(fid)
     98             if tag is None:
     99                 break  # HACK : to fix file ending with empty tag...

/home/umr9752/.local/lib/python2.7/site-packages/mne-0.9.dev-py2.7.egg/mne/io/tag.pyc in read_tag_info(fid)
    145     tag = Tag(*struct.unpack(">iiii", s))
    146     if tag.next == 0:
--> 147         fid.seek(tag.size, 1)
    148     elif tag.next > 0:
    149         fid.seek(tag.next, 0)

IOError: [Errno 22] Invalid argument

```

I am not sure how to proceed now...

dengemann, 2015-03-12:
cc @agramfort @mluessi

kingjr, 2015-03-12:
Actually the problems seems to only appear when I concatenate many epochs.

e.g. I concatenate without any problems epochs 1 to 5 without any problem. I did the same thing for 6 to 10: still no problem. But 1 to 10: and the saved file becomes unreadable..

dengemann, 2015-03-12:
> e.g. I concatenate without any problems epochs 1 to 5 without any problem. I did the same thing for 6 to 10: still no problem. But 1 to 10: and the saved file becomes unreadable..

That's interesting .... I need to think about it....

agramfort, 2015-03-12:
does your file become larger than 2GB?

kingjr, 2015-03-12:
yes

agramfort, 2015-03-12:
then you hit the size limit of fif files.

either use hdf5 or less epochs

kingjr, 2015-03-12:
Ok, super; thanks

A warning at the writing and the reading stage would help.

closed 2015-03-12 by kingjr

agramfort, 2015-03-12:
PR welcome

kingjr, 2015-03-12:
Shall I leave the issue open in the meantime then?

Eric89GXL, 2015-03-12:
I'll open a clearer bug description

kingjr, 2015-03-12:
ok thanks

agramfort, 2015-03-12:
create a more explicit issue with a gist to reproduce the problem