mne-tools/mne-python #1850: BUG: 2GB limit can be exceeded

Eric89GXL, 2015-03-12:
...on writes, which can leave us with un-readable files on read. See e.g. #1849.

Eric89GXL, 2015-03-12:
To recreate the problem (note: creates a huge file in memory and on disk):
```Python
import numpy as np
import mne

n_ch = 64
n_times_epo = 1024
n_needed = (2 ** 31) // (4 * n_ch) + n_times_epo  # make sure we go over
n_epochs = n_needed // n_times_epo
info = mne.create_info([str(x) for x in range(n_ch)], 1.)
raw = mne.io.RawArray(np.zeros((n_ch, n_times_epo), np.float32), info)
events = np.tile(np.atleast_2d([0, 0, 1]), (n_epochs, 1))
epochs = mne.Epochs(raw, events, 1, 0, n_times_epo - 1)
epochs.save('test-epo.fif')
epochs = mne.read_epochs('test-epo.fif')
```
Result:
```Python
  File "/home/larsoner/.local/lib/python2.7/site-packages/mne/io/tag.py", line 147, in read_tag_info
    fid.seek(tag.size, 1)
IOError: [Errno 22] Invalid argument
```

Eric89GXL, 2015-03-12:
@agramfort would it be good enough for now to add a check to the FIFF-ending function that checks the file size and throws an error if the limit is exceeded?

agramfort, 2015-03-12:
maybe :)

PR #1851 (2015-03-12): FIX: Check for 2GB limit
Closes #1850.

@agramfort see if you're happy with this.
closed 2015-03-12 by Eric89GXL