mne-tools/mne-python #1853: Q: log scale in viz.plot_image_epochs

kingjr, 2015-03-13:
I am trying to plot the trigger channel across epochs to check that it's adequately aligned using : 
```mne.viz.plot_image_epochs(epochs, trigger_ch, scalings=dict(stim=4096), units=dict(stim=''))```

I often generally trigger values in a single channel using a binary coding, which thus vary in an ~ exponential fashion (1 2 4 8 16 32 64 128 ... ).

I would thus like to plot them on a logarithmic colorscale. I thought of either 
- directly changing the epoch._data, 
- adding an option in mne.viz.plot_image_epochs
- or define a logarithmic colormap.
- [Recoding the values is not applicable in my case as some triggers can combine and thus produce values different from a power of 2.]

Any thoughts on the best strategy in this case?

agramfort, 2015-03-13:
+1 for

   - adding an option in mne.viz.plot_image_epochs

Famguy, 2016-01-03:
@agramfort From what I understood from the issue description, we need a new parameter, say `logscale` (set to `False` by default) in the `plot_epochs_image` function from [this](https://github.com/mne-tools/mne-python/blob/master/mne/viz/epochs.py#L28) file. 
Setting this parameter to `True` would convert the picks to logscale and give the corresponding plot. Please do correct me if I'm wrong.

jasmainak, 2016-01-03:
@Famguy picks is meant for selecting a subset of sensors in the data. You get the data using epochs.get_data(). You will also need to add tests, but make the PR first

agramfort, 2016-01-03:
@Famguy I would start by trying to replicate the use case of @kingjr to understand the pb better.

kingjr, 2016-01-03:
My use case was to check my triggers, which could vary between 0 and
>2**10, often with discontinuities (1-32, 64-128, 800-1024 etc);

To do this, I just manually changed the `epochs._data` with an `-np.log10`.

I am not sure that this is such a common case that it needs to be
implemented in mne.

agramfort, 2016-01-03:
can you share a image of how it renders with sample?

kingjr, 2016-01-03:
Sure:

```
import numpy as np
import matplotlib.pyplot as plt
import mne
from mne.datasets import spm_face

# Preprocess data
data_path = spm_face.data_path()
raw_fname = data_path + '/MEG/spm/SPM_CTF_MEG_example_faces1_3D_raw.fif'
raw = mne.io.Raw(raw_fname, preload=True)
epochs = mne.Epochs(raw, events=mne.find_events(raw, stim_channel='UPPT001'),
                    event_id={"faces": 1, "scrambled": 2}, tmin=-0.1, tmax=0.5,
                    picks=[0], baseline=None, preload=True, verbose=False)
# make synthetic trigger
epochs._data *= 0
t_stim = np.where(epochs.times > 0)[0][0:10]
t_resp = np.where(epochs.times > .400)[0][0:10]
for trial in range(len(epochs)):
    # typical stimulus trigger
    epochs._data[trial, 0, t_stim] = 2 ** np.random.randint(0, 5)
    # typical response trigger
    epochs._data[trial, 0, t_resp] = 2 ** np.random.randint(5, 10)

# log plot
fig, ax = plt.subplots(1)
im = ax.matshow(np.log2(1 + epochs._data[:, 0, :]), cmap='viridis')
plt.colorbar(im, ax=plt.gca())
ax.set_xlabel('Times')
ax.set_ylabel('Epochs')
plt.show()
```

[image]

jona-sassenhagen, 2016-04-23:
This can be closed I think @kingjr ?

kingjr, 2016-04-23:
Yes

jona-sassenhagen, 2016-04-24:
So close :)

closed 2016-04-25 by kingjr