mne-tools/mne-python #1862: Interpolate bad channels per epoch

christianbrodbeck, 2015-03-15:
I would like to interpolate bad channels on a per epoch basis. I noticed #1777 has the same desired feature but does not seem to have an efficient implementation yet, so it seems like a partial feature that I could work on. I would like as a basis something like a list of lists of bad channels i.e. something like:

```py
epochs.interpolate_bads([["CZ"], [], [], ["F1", "F3"], ...])
```

What do you think?

agramfort, 2015-03-15:
Hum... I am not sure I get the use case. You have a denoising problem?

christianbrodbeck, 2015-03-15:
Besides @wmvanvliet 's use case for FASTER I also have a lot of epochs in my EEG data where a single channel acts weird, even though the same channel is perfectly normal in the majority of the other epochs of the rest of the experiment.

wmvanvliet, 2015-03-15:
I was the same thing in many of my EEG datasets when working on FASTER. Manually going through the epochs and selecting bad channels is a pain though... but with some outlier detection, this interpolation of bad channels can really save some terrible datasets. I'm all for adding this functionality if @christianbrodbeck is up for it.

dengemann, 2015-03-15:
Why don't we add a parameter to the existing function that allows to interpolate epoch wise? But that would only be for epochs then. Maybe separate method or function would be cleaner then.
And yes, we need this for FASTER.

christianbrodbeck, 2015-03-15:
@dengemann that would mean a separate `InterpolationMixin` for Epochs, or (since it would be used only once) just adding the corresponding `interpolate_bads_eeg(self, channels_by_epoch=None)` function to `_BaseEpochs`, right? If you're ok with that I will start working on a PR.

A related question: why is it not allowed to interpolate MEG channels? Does this have something to do with not knowing how to combine grad and mag information, and if so would it be valid for us KIT users with only mag to interpolate bad channels? That would also save epochs for us.

dengemann, 2015-03-15:
> 
> @dengemann that would mean a separate InterpolationMixin for Epochs, or (since it would be used only once) just adding the corresponding interpolate_bads_eeg(self, channels_by_epoch=None) function to _BaseEpochs, right? If you're ok with that I will start working on a PR.
Wait. The API would the take a list of bads as input? 
> A related question: why is it not allowed to interpolate MEG channels? Does this have something to do with not knowing how to combine grad and mag information, and if so would it be valid for us KIT users with only mag to interpolate bad channels? That would also save epochs for us.
I think we discussed this, it was the grad / mag and the helmet VS head thing issue if I remember correctly. The good way to deal with it for MEG would be based on the forward model as in the fieldlines example.

christianbrodbeck, 2015-03-15:
@dengemann my API proposal was an optional list of lists, like

```py
>>> epochs.interpolate_bads_eeg([["CZ"], [], [], ["F1", "F3"], ...])
```

Can you (or anyone) think of a better option?

agramfort, 2015-03-15:
How would you get this list of list in practice? I feel it exposes a complex API and that could be an internal thing in a high level denoising function

Regarding interp of Meg the issue is that you have a vector field and you need to interpolate using sensor orientation and maybe even geometry

christianbrodbeck, 2015-03-15:
> How would you get this list of list in practice? 

FASTER supposedly would generate the list automatically. I have a (wxpython and OS X dependent) GUI  that can create such a list. I assume there could be other ways of generating the list in the future, it seemed like the most obvious representation of the information.

christianbrodbeck, 2015-03-15:
> I think we discussed this, it was the grad / mag and the helmet VS head thing issue if I remember correctly. The good way to deal with it for MEG would be based on the forward model as in the fieldlines example.

I can't find a relevant open issue (searching for "interpolate MEG" and such) is there one or should I create a new one?

agramfort, 2015-03-15:
I would say let's wait for Faster to be ready then.

agramfort, 2015-03-15:
> I can't find a relevant open issue (searching for "interpolate MEG" and
such) is there one or should I create a new one?

feel free to create it.

christianbrodbeck, 2015-03-15:
> I would say let's wait for Faster to be ready then.

@agramfort I need the feature now and I would be willing to work on it. Are you saying because my current use case is not everyone's use case you don't want to have the feature added before another use case is implemented?

dengemann, 2015-03-15:
Can we handle this using a private function? Christian could use it immediately, the mixin could use it, finally we can decide later about the public API.

christianbrodbeck, 2015-03-16:
That would work for me

agramfort, 2015-03-16:
Give it a try in a branch and WIP PR to make this public and let's talk about merge when faster is ready

christianbrodbeck, 2015-03-16:
closing for #1869

closed 2015-03-16 by christianbrodbeck

christianbrodbeck, 2015-03-17:
I got it working, here is an illustration of the problem (see the blue dotted channel in the top left plot and the topomap).

[image]

 By the way if anyone is interested in the GUI let me know, these are the limitations:

- requires wxPython
- only one modality at a time GRAD, MAG or EEG
- adapted to OS X, not sure what it would take to make it run smooth on other platforms