mne-tools/mne-python #1868: EHN: add tmin and tmax key to evoked

kingjr, 2015-03-16:
Is there any reason not to pass the tmin and tmax keys of epoched data to evoked data

e.g.
```
evoked = epochs.average()
print(evoked.tmin)
```

Thanks

dengemann, 2015-03-16:
> Is there any reason not to pass the tmin and tmax keys of epoched data to evoked data

Probably to match the MNE-C FIFF defnition of the Evoked structure or because of lazyness.
One coul add these attributes on construction / read time.
You can do `print(evoked.times.min())`.

I perosnally don't really care. I think I've never used these attributes.

agramfort, 2015-03-16:
A bit of a YAGNI to me

Or maybe via property

kingjr, 2015-03-16:
It's no more necessary to have it in epochs, as users could also do epochs.times.tmin. I would add it for sake of coherency.

agramfort, 2015-03-16:
Tmin in epochs is used to read from raw and to define times. Once you have times then tmin and tmax are not required

kingjr, 2015-03-16:
Faire enough.

closed 2015-03-16 by kingjr