mne-tools/mne-python #1872: recursive travis?

dgwakeman, 2015-03-17:
I'm not sure if this is definitely an issue or not in the way travis is setup with appveyor. I noticed on my last commit, and it seems to be starting the same way now...

When I push:
Travis starts and runs till completion. I can see the results when I click on the details link
Then appveyor starts and runs till completion
then Travis starts and runs till completion. Only then does the github page result get updated.

Is this right? @Eric89GXL

Eric89GXL, 2015-03-17:
No, they run on their own. Sometimes the github page won't update its status, but if you refresh your browser it should.

Eric89GXL, 2015-03-17:
Also note that sometimes CI builds are started or stopped manually. For example, if I see that an AppVeyor build has failed because it didn't download the testing dataset, I'll often restart it and that fixes the problem. Or if I see that someone has push-forced replacing a recent commit, I'll stop the old Travis build to save Travis some work and to get us feedback faster.

But I think what you're seeing is a GitHub page updating issue more than anything else. Note that you can also get status switches if you push a new commit while the old one is building -- it will go from potentially completed to in progress again. Also, if I navigate to Travis and then click "back" to get to the PR page in Chrome, often the page reverts to the first version Chrome loaded, which is outdated because it doesn't take into account the on-the-fly page updates it received.

I'm going to go ahead and close this, feel free to open if you continue to see suspicious things that can't be accounted for by GitHub / Chrome / Firefox just not updating the page appropriately.

closed 2015-03-17 by Eric89GXL

dgwakeman, 2015-03-19:
This is doing it right now. I just got an email from travis saying that my build is bad. Linking here:
https://travis-ci.org/dgwakeman/mne-python/builds/55035004
But when I click on the link on github for the latest commit it hasn't run yet.

dgwakeman, 2015-03-19:
open 1872

Eric89GXL, 2015-03-19:
That build is running for `dgwakeman/mne-python`, not `mne-tools/mne-python`. You can turn off builds for your own repo if you want.

dgwakeman, 2015-03-19:
Ahhh. ok Thanks sorry

Eric89GXL, 2015-03-19:
No problem, that's an easy one to miss :)