mne-tools/mne-python #1875: Bug: Reading Evoked data from save -ave.fif files

Anevar, 2015-03-18:
There seems to be problem(a bug) with reading the saved evoked data (-ave.fif files) using  mne.read_evokeds. 

The evoked data is present before we save it into a file but there seems to be a problem when we read it. (more specifically-> there are no data present for some trails when we read it)

As a result the average of subject over trials or the grand average could not be obtained. 
I use mne version '0.9.dev'.

agramfort, 2015-03-18:
can you share the file that creates the problem as well as a script to
produce the error?

Anevar, 2015-03-18:
Yes I can send them to you by mail as I'm not able to send in the file on github.

Anevar, 2015-03-18:
Dear Alex,

I'm attaching the script along with this email and result of the average
over trials for each subject.

The evoked data is present before I save it unfortunately for a few runs it
was only NaNs when I read it.(hence the average is NaN)
I'm not sure if the evoked.save method is introducing errors or its the
read_evokeds.

There is also a problem when I tried to append the evoked of the all the
subject for all runs to a list so that I can analyse each subject
individually. When I did this, the evoked data (60 channels x 1201 )
appends as evoked data of shape ( 1 x 1201). Hence resorted to extracting
the data, time information separately inside the def ern() in the script.

If you are connected to neurospin, you can find the script under the link,

'/neurospin/meg/meg_tmp/erntime_clemence_2012/Hafeza/Scripts/'

FYI: read_evok_avg_plot script reads the evoked file, computes and plots
the average for each subject and grand average. also plots the evoked data,
topomap.

Best Regards,
Hafeza

agramfort, 2015-03-18:
please share a tiny script that reproduces the error not your full script

also please share the broken -ave.fif file via dropbox or google drive. I
am not at Neurospin nor at work.

agramfort, 2015-03-19:
ok. I got the problem.

can you read the original raw file?

problem occurs at line 90 in proc_history.py is due to the fact that
this a non ascii character in user name (é)

cc @EricGXL89

agramfort, 2015-03-19:
also don't hesitate to share the files with @Eric89GXL and please add a raw file to drive so we can debug as I suspect we don't write back correctly after reading the raw file.

Eric89GXL, 2015-03-19:
Tomorrow I can try to reproduce it by putting some non-ascii chars in an evoked `info` and writing/reading it

agramfort, 2015-03-19:
great idea. Thanks for taking a look

PR #1877 (2015-03-19): FIX: Unicode
Closes #1875.

@agramfort ready for merge from my end assuming the CIs are happy. `test_meas_info.py` passes on my 3.4 and 2.7 so I'm optimistic this might be a one-commit change.
closed 2015-03-19 by agramfort