mne-tools/mne-python #1878: BUG: EGI file reading won't work if no events are present.

fraimondo, 2015-03-19:
When reading a RAW EGI file using `read_raw_egi`, if no events are present, then the function fails with `list index out of range` in `_RawEGI` `__init__`.

Digging into the file around line 203:
```Python
event_codes = list(egi_info['event_codes'])
egi_events = data[-egi_info['n_events']:]
```
With no events, `event_codes` is an empty array, while `egi_events` results in the last data row.

Later in the file, the line that gives us this beautiful error message.
```Python
for ii, event in enumerate(egi_events):
    if event.sum() <= 1 and event_codes[ii]:
        more_excludes.append(ii)
```

I can fix it, just wanted to check the status before.

CC: @dengemann

agramfort, 2015-03-19:
Can you open a PR?

fraimondo, 2015-03-19:
I'm on it.

closed 2015-04-10 by agramfort