mne-tools/mne-python #1882: FIX: MNE-Report section ordering broken if using custom sections

dengemann, 2015-03-20:
that overlap with the SECTION_ODER list in mne.report.py

The reordering code above and in [this line](https://github.com/mne-tools/mne-python/blob/master/mne/report.py#L1090) seems pretty broken to me btw. It also changes the state of the report, after saving the order in report.section and report._sectionlabels is changed.

cc @mainakjas

mainakjas, 2015-03-20:
I agree this part could be improved. Can you provide a small script to illustrate the problem?

dengemann, 2015-03-22:
@mainakjas toy example 

```Python
import matplotlib.pyplot as plt
from mne.report import Report
report = Report()
fig1 = plt.figure()
plt.plot(1, 4)
fig2 = plt.figure()
plt.plot(2, 3)
fig3 = plt.figure()
plt.plot(4, 5)
report.add_figs_to_section(fig1, 'test-figure', 'test-section')
report.add_figs_to_section(fig2, 'test-figure2', 'evoked')
report.add_figs_to_section(fig3, 'test-figure4', 'test-section2')
report.save('test-report.html')
```

mainakjas, 2015-03-27:
hmm ... this is totally weird. It used to work as far as I remember. I'll look into it tomorrow.

teonlamont, 2015-04-12:
hey @mainakjas, do you think this could be resolved for 0.9?

mainakjas, 2015-04-12:
okay, I think I figured out what the problem was @dengemann 

When we run the report through the command line, we want to have the sections ordered as: raw -> events -> epochs -> evoked -> covariance -> trans -> mri -> forward -> inverse

It so happened that one of your section names was exactly one of these pre-defined section names, i.e., `evoked`. If you change that, you should be good. We can throw an error and not allow the user to enter one of these section names. Or give a warning that it will mess up the order, whichever you prefer.

agramfort, 2015-04-12:
to me it should just work :)

dengemann, 2015-04-12:
I know mainak :) I just wanted to be able to have it working even if I call a section 'evoked'. I'm -1 for the warnings approach, please let's have some magic here. If we're in scripting mode, we should not stick with these defaults.

mainakjas, 2015-04-12:
Can't you call it `Evoked`? Just want to avoid a code mess :)

mainakjas, 2015-04-12:
I can try to make it work if you insist and have a good reason for it :)

mainakjas, 2015-04-12:
... or maybe I really need to revisit the hashing function. We always keep coming up with new section names for which the report breaks.

dengemann, 2015-04-12:
To me we should maybe detach scripting from parsing mode. I don't want any of these defaults when I use the Report in my code.

closed 2015-04-17 by Eric89GXL