mne-tools/mne-python #1883: reject trials based on head movements

agramfort, 2015-03-20:
we should have an option to reject epochs based on head motion tracking. If too strong movement -> reject.

SPM has something like this. See spm_eeg_megheadloc.m

Eric89GXL, 2015-03-20:
With #1876 we can even do it on non-Maxfiltered data if cHPI is used. Although I doubt there are many people using cHPI without using Maxfilter...

agramfort, 2015-03-20:
is this PR ready for merge?

would it make sense to be able to plot motion params over time? it's quite
common for fMRI.

Also there is this old issue to reproduce mne_map_data to correct head
motions between runs without using maxfiler. Could it be part of GSOC?

Eric89GXL, 2015-03-20:
Nope, not ready. I have only tested it with simulated (infinite SNR) data, and haven't checked the params. I'm not even sure if the rotation matrices are correct, I only checked the positions so far...

I plan to work on it in early April, I'd project it'll be ready for merge by the end of April. But in any case, the functionality mentioned here should be able to be implemented using the existing `get_chpi_positions` for any MaxFiltered file, since MaxFilter records and stores the extracted head positions in the output _raw_sss.fif file. #1876 will just extend it work with files that have not been processed with MaxFilter but have been recorded with cHPI on.

agramfort, 2015-03-20:
ok. Let's not forget.

Eric89GXL, 2017-02-16:
See #3979 for suitable solution

closed 2017-02-16 by Eric89GXL