mne-tools/mne-python #1887: FIX: MNE-Report makes unrequested copies of section and memory blows up if integers are used in sections

dengemann, 2015-03-22:
@mainakjas 

```
import matplotlib.pyplot as plt
from mne.report import Report
report = Report()
fig1 = plt.figure()
plt.plot(1, 4)
fig2 = plt.figure()
plt.plot(2, 3)
fig3 = plt.figure()
plt.plot(4, 5)
report.add_figs_to_section(fig1, 'test-figure', '2347')
report.add_figs_to_section(fig2, 'test-figure2', '2722')
report.add_figs_to_section(fig3, 'test-figure4', '1735')
report.save('test-report.html')
```

In this case self._sectionvars will be a dict of sections pointing to an empty string. This probably creates all the trouble. The problem is in _clean_varnames:

In [39]: _clean_varnames('1334')
Out[39]: ''

This is not really desirable. Can we avoid this or is there a reason integer strings are not allowed? Often subjects just have such integer codes.

dengemann, 2015-03-22:
If it cannot be changed it should be handled either with an error or by adding characters to the string ...

dengemann, 2015-03-22:
I think the evil comes when section labels are a list of empty strings... then the labels are not unique any more and figures get added everywhere on saving.

mainakjas, 2015-03-27:
okay, got it. I could reproduce the problem. I'll work on it tomorrow. Maybe we should reconsider @agramfort 's idea of using python's hash function?

agramfort, 2015-03-27:
whatever works !

closed 2015-04-19 by teonlamont