mne-tools/mne-python #1888: plot_topo cosmetics

jona-sassenhagen, 2015-03-22:
mne.viz.plot_topo seems a bit half-assed right now, unless I'm missing something. However, when suggesting a student of mine to use MNE for a basic ERP analysis, I realised for people who'd use MNE mainly for plain old boring ERPs, this is probably an essential feature. I'm currently adding all the minor cosmetic things necessary for basic usage:

- a legend
- an option to set linewidth so plots don't look like crayon drawings
- plot tickmarks and spines
- auto set complimentary label and background color
- channel names

And so on. Is this a good idea, or am I just duplicating stuff already done better elsewhere? I'll submit a PR if you think it's a good idea.
Example of status quo:

[image]

agramfort, 2015-03-22:
a very welcome contribution !

mainakjas, 2015-03-22:
yep, I have wanted the legends too :) I write my custom scripts for that but it would be great if it can be integrated into MNE

dengemann, 2015-03-22:
Sounds great!

teonlamont, 2015-04-12:
closing for pr https://github.com/mne-tools/mne-python/pull/1890.

closed 2015-04-12 by teonlamont